ControlSystemStudio/phoebus #2073: What is the best way of setting a macro at runtime?

lorenzo-gomez-windhover, 2021-11-29:
Hi there,

Hope you are all doing well.

I have a display which has a macro called `CPUID`. Initially the value of this macro set on the builder is "CPU1". I would like a script to change the value of that macro at runtime by clicking a button.

I tried the following in a script that is attached to a `Action Button` and it didn't work:
`widget.getTopDisplayModel().getEffectiveMacros().add("CPUID", "CPU2")`

I thought since `getEffectiveMacros` is just a hashmap, it would update the macro after I click on "Reload Display". But that did not happen. The macro(which is being used by a label on the same OPI) keeps the same value of "CPU1". I looked around in documentation and issues and did not find anyone who has tried updating macros this way. I wonder if this is by design?

Is there a possibility of changing macro values at run time? I would be ok with the user having to manually click on "Reload Display". But my team would really like to do this by clicking a button on the OPI.

Hope this is clear enough.
Thanks in advance 
Lorenzo

lorenzo-gomez-windhover, 2021-11-29:
I think I may answer my own question, silly me. It's just like any other property, right? I'm assuming it is something like `widget.setProperty("macros", new MacroObject())`, right?

kasemir, 2021-11-29:
Macros are optimized to only be evaluated when opening a display. We don't check for macro changes at runtime.

If you want to change macros at runtime, you can do that with an embedded display, but you also need to change the display file to force a re-load.

If all else fails, study the examples. See examples -> embedded, https://github.com/ControlSystemStudio/phoebus/tree/master/app/display/model/src/main/resources/examples/embedded, this detail in update_display.py:

```
    # Next value: Show same file with different macros
    widget.getPropertyValue("macros").add("M", "Value 2")
    # Need to change the file name and then revert back to force a reload
    widget.setPropertyValue("file", "")
    widget.setPropertyValue("file", "a.bob")
```

lorenzo-gomez-windhover, 2021-11-29:
I see. Thanks a lot for the info! I'll explore this option further at some point.

lorenzo-gomez-windhover, 2021-11-29:
I'm just curious; do you guys ever intend to add the capability for changing the value of a macro at runtime?

closed 2021-12-01 by lorenzo-gomez-windhover