ControlSystemStudio/phoebus #2086: Critical vulnerability in Apache Log4j library CVE-2021-44228

pavel-ch, 2021-12-13:
CVE:
https://cve.mitre.org/cgi-bin/cvename.cgi?name=CVE-2021-44228
Info for example:
https://www.kaspersky.com/blog/log4shell-critical-vulnerability-in-apache-log4j/43124/
Do we have to react?

georgweiss, 2021-12-13:
Spring Boot-based services will depend on the problematic log4j lib. Looking into it now.
In the meanwhile, services can be restarted adding the following to the command line:
``-Dlog4j2.formatMsgNoLookups=true``

pavel-ch, 2021-12-13:
Now I see version
log4j-api-2.11.1.jar, we probably should upgrade to 2.15?

georgweiss, 2021-12-13:
I have investigated how to do this in the save-and-restore service. Unfortunately not only a matter of changing a version number. But I have a working pom that will pull in 2.15.0.
Will check other services later today.

pavel-ch, 2021-12-13:
In the Phoebus product libs, I see the log4j-api-2.11.1.jar and log4j-to-slf4j-2.11.2.jar. According to information I have found, these jars are fortunately NOT subject of this CVE (that is for log4j-core module).

georgweiss, 2021-12-13:
I have concluded the same.
Will check how to handle product libs.

pavel-ch, 2022-06-17:
Closing now but opening a new ticket regarding general dependency vulnerabilities

closed 2022-06-17 by pavel-ch