ControlSystemStudio/phoebus #2090: Replace TextField by TextArea for PV list Name edition in app-alarm-ui and core-ui

katysaintin, 2021-12-15:
It will be more convenient to use a TextArea instead of TextField for listing the PV name to add in Alarm Service

Code to change  : 
- app-alarm-ui org.phoebus.applications.alarm.ui.tree.AddComponentAction.java
  Line 45 replace private final TextField name = new TextField(); by private TextArea name = new TextArea();
- core-ui : org.phoebus.ui.autocomplete.AutocompleteMenu.java 
  Line 113 replace  TextField by TextInputControl (TextInputControl is the mother-class of TextField and TextArea)

georgweiss, 2021-12-17:
Feel free to create a pull request that we can approve.

katysaintin, 2021-12-17:
> 

I try to commit my suggestion in my own branch CSSTUDIO-2090 but I'm not authorized. Do you know if I need specific  credentials to contribute ? Thank you (I use my login katysaintin and my github password)

kasemir, 2021-12-17:
I don't think there are any restrictions on using github to fork, commit to your fork, then create a PR

kasemir, 2022-01-27:
Since this change, on our linux systems in the control room, we can no longer enter PV names.
For" Type: Node" selected, you can enter text into the text area. For "Type: PV/s", pressing any key opens the PV name completion drop-down, and you can no longer enter anything.
For other plain text fields like in Probe the PS name completion drop-down still allows editing the PV name, so the issue is somewhere in the text area and/or how the alarm's "Add .." dialog uses it.

georgweiss, 2022-01-28:
I'm seeing the same issue on MacOS.

katysaintin, 2022-01-28:
> I'm seeing the same issue on MacOS. The problem occurs in AutoCompleteMenu L91 which assumes that the input component is a TextField. I changed the cast to TextInputControl, which fixes the issue.
> 
> Will submit a PR once I have done some more testing.

Yes you are right , I have miss this one. Thank you for your correction, tell me if you have no time to do it.

georgweiss, 2022-01-28:
Already done, pending in #2118

kasemir, 2022-01-28:
Thanks, that was quick!

closed 2022-01-28 by kasemir

georgweiss, 2022-01-28:
Well, I'm about to release a new version at ESS...