ControlSystemStudio/phoebus #2094: [Question] units on local pvs

jjaraalm, 2021-12-16:
Is there any way to set units on a local pv like `loc://foo<VDouble>(0)`?

I tried using a script to manually create a `VDouble` with a `Display` and assign it. This works up until I try to interact with the widget (e.g., incrementing a spinner) and then the units disappear.

crispd, 2022-03-02:
Is it an option to make an EPICS PV?

kasemir, 2022-03-03:
Yes, I'd say that's the way to go.
Local PVs are not meant to duplicate all those PV attributes. The syntax would explode into something like `loc://xxx(2, units=..., prec=..., ......)`. Local PVs are meant for standalone demonstrations or tests of the tools. They have little use in an operational display that interacts with the actual control system.
With a IOC PV you can set hard limits, display limits, units, precision and much more. You don't have to modify any existing IOC for that, you can always add such PVs that are for displays into a new soft IOC:

```
# Run as
#  softIoc -d name_of_file.db
record(ao, "foo")
{
  field(DOL,  "42")
  field(PINI,  "YES")
  field(DRVH, "100")
  field(DRVL, "-100")
  field(DRVH, "90")
  field(DRVL, "-90")
  field(EGU,  "mm")
  field(PREC, "2")
# Could add alarms, MDEL, ADEL, ...
}
```