ControlSystemStudio/phoebus #2103: CS-Studio title

kasemir, 2022-01-20:
We started out with a window title of "Phoebus". In Nov. 2019 we changed it to "CS-Studio (Phoebus)" to indicate that this is CS-Studio in its latest incarnation.

https://github.com/ControlSystemStudio/phoebus/blob/7f3e17d539f552475b008143d073d5d2e433588b/core/ui/src/main/resources/org/phoebus/ui/application/messages.properties#L33

In places where I only run "CS-Studio (Phoebus)", some users are still confused as to when they can get the "real CS-Studio", not this Phoebus thing. Is it time to change the title to just "CS-Studio"?

Kunal @shroffk , do you rely on concurrently running both the Eclipse-based and new CS-Studio, and need a way to distinguish them? Would it be possible to do that by other means, still allowing plain "CS-Studio" for the new window title?

georgweiss, 2022-01-20:
OK with me to change.

kasemir, 2022-01-20:
Is ESS still using the Eclipse-based version? .. concurrently on the same host with the new one? Or all moved up to the new one?

georgweiss, 2022-01-21:
We have a few users of the Eclipse-based version, and that is for one sub-system. They are currently migrating, so later this year it's Phoebus (sorry, CS-Studio) only.

closed 2022-03-01 by shroffk