ControlSystemStudio/phoebus #2127: "Open Display" action does not work as expected from within "Template/Instance" widget

Amadrath, 2022-02-08:
Hi everyone,

if I call an "Open Display"-action from within an instance, it fails because the widgets uses the directory of the Phoebus-Installation as reference for the relative path, instead of the directory of the display file. The same action works fine, if used from a display.
I assume, that the Template/Instance widget does not inherit the root directory from the display, it is part of. Is that intended? Is there a way, to resolve that?

kasemir, 2022-02-08:
The template/instance widget is the most recently added widget, relatively new, and this is likely a bug in there that needs to be fixed.
Can you provide a simple example display that demonstrates the issue, that'll help in looking into this in more detail.

Amadrath, 2022-02-08:
Here you go:
[DEMO.zip](https://github.com/ControlSystemStudio/phoebus/files/8024869/DEMO.zip)
It's three files, one display with the Template/instance widget, one display, that is called by the instance widget and one display, that is to be opened. Simply right-click on the text-widgets, and choose the "Open Plot"-action. It works fine on the non-instanced widget, but generates an error on the instanced ones.

commit 08e0b701 (2022-02-08): Template/instance widget: Pass display path info down to content ...

.. to allow resolution of relative path names

Fixes #2127
closed 2022-02-08 by kasemir