ControlSystemStudio/phoebus #2145: PVAFloatArray setValue error

joshikk, 2022-02-21:
I am having a float array in python pvaccess (PvScalarArray(FLOAT)). when I am modifying any element of the array I am getting a following error in phoebus. This is not so if the array is of PvScalarArray(STRING).

```
2022-02-21 10:27:38 WARNING [org.epics.pva] TCP sender /[10.15.98.150:5075](http://10.15.98.150:5075/) request encoding error
java.lang.Exception: Cannot set float[] value to [D@1c7da38b
        at org.epics.pva.data.PVAFloatArray.setValue(PVAFloatArray.java:73)
        at org.epics.pva.client.PutRequest.encodeRequest(PutRequest.java:138)
        at org.epics.pva.common.TCPHandler.sender(TCPHandler.java:179)
        at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
        at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1130)
        at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:630)
        at java.base/java.lang.Thread.run(Thread.java:831)
```

kasemir, 2022-02-21:
Looks like you're mixing libraries.
The error message is from the new Java PVA client, https://github.com/ControlSystemStudio/phoebus/tree/master/core/pva
The PvScalarArray that you're trying to pass in is from the older epics-pvdata library.

--> Stick to one. If you use the new one, types you can 'put' include `PVAFloatArray`, `float[]`, `List<Number>`, see https://github.com/ControlSystemStudio/phoebus/blob/b6bc5343399045fb21cafb5ed4562d26e106c8a2/core/pva/src/main/java/org/epics/pva/data/PVAFloatArray.java#L48

joshikk, 2022-02-22:
I am using the pvaccess (pvapy 4.1.0) server on python, It is independent
of the EPICS installation
There is no PVAFloatArray listed in the documentation of pvapy.
If I understand correctly, the problem is that pvapy is using the old
epics-pvdata library, or the documentation is not uptodate??

joshikk, 2022-02-22:
PS: I am not having any script on phoebus.
I have just placed the array control embedding the textupdate control on
the main panel.
For String, the pvaccess is working but not for float array.

How do I make the controls use the old library calls, without scripts??

kasemir, 2022-02-22:
OK, that makes it more obvious what you have:

 * Python-based PVA server for array of string or array of float
 * Display that uses text entry? text update? within array widget

String array works, float array runs into that error.
Since you already have a running test setup, can you try double and int arrays?

joshikk, 2022-02-24:
It is working for DOUBLE data type array
However for INT and UINT data type array, phoebus is reporting similar
error ie. with PVAIntArray

kasemir, 2022-02-24:
Got it. The array widget simplifies to either a double or string array, which works well enough for display.
Writing back to channel access, the PV support also allows writing for example a double array to a float array.
The PV access implementation currently is stricter. I'll add some more type conversions there so it's more like the channel access PVs, since this will likely come up again.

kasemir, 2022-02-24:
There's a pull request that should fix it. Didn't test with python-based PVA server, but used this type of record with softIocPVA from R7.0.6:
```
record(waveform, "array")
{
    field(FTVL, "FLOAT")
    field(NELM, "10")
    field(INP, {const:[65, 66, 67, 68, 69]})
    field(PINI, "YES")
}
```

and checked that a display with "Array" widgets containing either a Text Update or Text Entry can read/write for FTVL=DOUBLE, FLOAT, INT64, LONG, SHORT, BYTE.

PR #2149 (2022-02-25): PVA: Allow setting all array types from double[]
Fixes #2145
closed 2022-02-25 by kasemir