ControlSystemStudio/phoebus #2168: org.phoebus.framework.autocomplete preferences not working

tynanford, 2022-03-09:
Our users really like the autocomplete feature in Phoebus. Since we aren't using PVA at this time, I'd like to remove that from the autocomplete suggestions.

These settings are available but setting them has no impact. 

https://control-system-studio.readthedocs.io/en/latest/preference_properties.html#framework-autocomplete

It looks to be a bug here: https://github.com/ControlSystemStudio/phoebus/blob/master/core/framework/src/main/java/org/phoebus/framework/autocomplete/PVProposalService.java#L27

Pull request with a possible fix is incoming but not sure if I went about it the right way

[image]

tynanford, 2022-03-15:
great thanks!

closed 2022-03-15 by tynanford

kasemir, 2022-03-15:
Thanks for finding this issue in the first place, makes a lot of sense to exclude the PVA suggestions at many sites.

PR #2174 (2022-03-15): PV Proposal services used wrong package for preferences
Fixes #2168 , much simpler than #2169