ControlSystemStudio/phoebus #2171: Multiple data browser issues

georgweiss, 2022-03-10:
Detected a number of data browser issues related to the plot config dialog:

1. If a title is specified in the dialog, it is rendered in the plot. But the title field in the Misc tab is not updated. The "show legend" value on the other hand is in sync in the Misc tab.
2. A change of trace color in the dialog will update the plotted trace, but the change is not reflected in the Traces tab.
3. A change of visibility is reflected in the plot, but not in the Traces tab.

shroffk, 2022-03-14:
The configuration dialog works to change the **runtime** settings of the plot. This can be quite useful when you want to make changes specific to your current environment ( display, device,...) but don't want the changes to be permanent. This is also very useful for plots that are embedded in Display builder screens.

But... I too have had users who are often confused and annoyed why their configuration changes are "lost"

While the dialog mentions runtime changes, we might want to elaborate on that a bit. Maybe in the dialog we can mention "these changes will not be persisted, use properties panel for making permanent modifications"

kasemir, 2022-03-14:
In an operational environment, the changes can't be saved because the display file is read from a read only disk location or fetched via http. That's on purpose so that one user zooming beyond a useful data change, disabling all traces or changing all colors to black won't affect the next user who opens the display.

But... yes. Maybe add "Runtime Settings" in the dialog title, and hide the longer "these changes will not be persisted, use properties panel for making permanent modifications" behind some (?) help button.

georgweiss, 2022-03-14:
But one may object that even for a read-only plt file the user may change colors from the trace tab, even though that cannot be saved back to the file. So in practice the tabs also offer means to change the runtime properties.

georgweiss, 2022-03-16:
I will update the dialog title, but I will put clarifications about how changes can or cannot be persisted in the documentation for the data browser.

georgweiss, 2022-03-17:
My users point out - and I do agree - that the subtle differences between the dialog and the tabs in the data browser are a bit too subtle to make sense in a day-to-day use case. Documenting these differences does not really help frustrated users.

My suggestion would be to completely disable the dialog in the context of the data browser. I'd be prepared to add a preference setting for that. 

Any thoughts @kasemir , @shroffk ?

kasemir, 2022-03-17:
I'm more than OK with a preference to disable the runtime dialog.
The runtime dialog only exists because BOY had one, and that one was very confusing because it allowed access to everything. For instance, you could change the "time axis" into a "log axis" and back, which doesn't make any sense because your data is either time based or not, the horizontal axis mode needs to be fixed. The current runtime dialog is already a much reduced version of what was in BOY, and I'm all for disabling it.

shroffk, 2022-03-17:
I am OK with a preference too.
I think I might actually disable it for most users and enable it for some experienced operations who would understand the difference.

georgweiss, 2022-03-17:
> ...disable it for most users

So the default value for the new property should be "disabled", right?

closed 2022-03-22 by georgweiss