ControlSystemStudio/phoebus #2180: github actions failing

shroffk, 2022-03-16:
```
Error:  Failed to execute goal org.apache.maven.plugins:maven-deploy-plugin:2.7:deploy (default-deploy) on project parent: Failed to retrieve remote metadata org.phoebus:parent:4.6.6-SNAPSHOT/maven-metadata.xml: Could not transfer metadata org.phoebus:parent:4.6.6-SNAPSHOT/maven-metadata.xml from/to github (https://maven.pkg.github.com/ControlSystemStudio/phoebus): Not authorized , ReasonPhrase:Unauthorized. -> [Help 1]
```

shroffk, 2022-03-17:
@lorenzo-gomez-windhover can you look into this

shroffk, 2022-03-18:
@lorenzo-gomez-windhover I believe `GITHUB_TOKEN: ${{ secrets.GITHUB_TOKEN }}` is not available in the docker container on which the tests are being run
We will have to find a way to make this available for the publish step to work

lorenzo-gomez-windhover, 2022-03-18:
Yes, I missed that! There is a way of doing it. I will get on it as soon as possible.

shroffk, 2022-03-24:
It doesn't seem to have worked with the above fix :(

lorenzo-gomez-windhover, 2022-03-28:
Yes, we should. Sorry about this guys. I'll be making a PR shortly to go back to CI without containers. I'll keep looking into these issues further in the future.

andersharrisson-ess, 2022-03-28:
Sorry, I've been swamped with other work and haven't been able to help out on this issue. Reverting the change was reasonable.

commit 21b4b3cd (2022-03-24): Merge pull request #2192 from WindhoverLabs/ci

Fix  #2180
commit e4fa3f9d (2022-03-24): Fix  #2180
closed 2022-08-10 by shroffk