ControlSystemStudio/phoebus #2186: Scale x-axis on XY-Plots

Amadrath, 2022-03-21:
Hi,

I have array PVs with 16384 values. Another PV holds the x-scaling for the arrays. Is there a simple way, to generate a 16384 value array with a linear gradient, defined in the second PV, to serve as x-values for the first PV? Maybe as a local PV? Could I access the indices of an array PV and multiply it with the gradient from the other PV for that purpose?

georgweiss, 2022-03-21:
You can multiply arrays element-wise using the formula function ``arrayMult`` if input arrays are of equal length. For the PV specify:
``=arrayMult(array1, array2)``

kasemir, 2022-03-21:
The XYPlot plots the value of a X,Y waveform records.
The data should already be in those records.
If the records don't contain the correct data, computing that data in the display is conceptually wrong because the display should just be that, the display, not the data provider.
If you need to do array computations on the IOC, check the http://htmlpreview.github.io/?https://github.com/epics-modules/calc/blob/R3-6-1/documentation/calcDocs.html 
If your IOC generates the actual data, you can display it in the XYPlot, check it with "caget" on the command line, display it in the next upcoming display tool. If you instead create the data in the display tool, you'll have to re-do that work for every other display tool.

kasemir, 2022-03-21:
If all fails, you can use an aSub record with gradient and scaling factors as inputs and your C code to generate the desired X waveform. aSub is part of EPICS base.

Amadrath, 2022-03-21:
I think another solution would be to provide an empty (soft) aai record and write the values into it with a little script. As the gradient is pretty static, and we already use autosave, that array would only need to be rewritten every few months. This is a bit of a dirty hack, but easier to implement.

kasemir, 2022-03-21:
> .. we already use autosave,

Sure, that works as well. Plus autosave is part of synapps, so looks like you already know how to get synapps modules going ;-)

Amadrath, 2022-05-06:
Thanks kasemir. As I see it, this issue is resolved. Thus I hereby close this ticket.

closed 2022-05-06 by Amadrath