ControlSystemStudio/phoebus #2190: Plot samples order sometimes wrong

georgweiss, 2022-03-23:
I have no procedure to reproduce this, but the issue here seems to be related to IOC connection "quality".

In some cases the plot samples may contain duplicates (same time stamp and value), but the repeated occurrence is at "wrong" position in the sample set. Consequently the sample set is not ordered on time stamp and the plot rendering will reflect this, see screen shot. Not sure how to fix this in the best way. Ordering the plot sample before rendering seems a bit expensive.

Any other suggestions?

[image]

georgweiss, 2022-03-28:
@kasemir, @shroffk, any thoughts?

If ordering samples is acceptable I will create a PR.

kasemir, 2022-03-28:
What you describe seems to be a special case. We don't generally see this, you mention that you can't reproduce it locally, either. Adding a `sort()` call to each rendering run would be too expensive.
Better first instrument the code to detect this issue. For example, add a check to `LiveSamples#add`
```
final int N = samples.size();
if (N > 0)
{
    final PlotSample last = samples.get(N-1);
    if (sample.getPosition().isBefore(last.getPosition()))
    {
        System.err.println("New live sample " + sample + " goes back-in-time from " + last);
    }
}
```
If the issue is that we for some reason add an existing sample (same time stamp and value) more then once, maybe because of a channel access reconnect that sends an old value, then the best solution would be to detect that and simply skip the duplicate.
If it's more complicated, then a binary search for the proper insertion point at the time of adding the sample would be better then blindly sorting all samples all the time.

georgweiss, 2022-03-28:
I have no procedure to reproduce it locally, but I have seen the issue at one occasion. It seems it is related to the IOC connection quality (for lack of better explanation).

Agree that sorting blindly would be too crude. Inserting a check when samples are added can be attempted.

kasemir, 2022-03-28:
> connection quality (for lack of better explanation)

Not sure what that might be.
A channel access subscription ("monitor") for a setpoint type of PV tends to get some initial value that's old, reflecting the time and value when the PV was last changed. After a network disconnect and then reconnect, we get the same old sample again. But the PV in your screenshot seems to change and generate a new time stamp every few seconds.
Is this with PV Access instead of channel access, and is this going via a gateway? Running a separate "camonitor" or "pvmonitor" in a terminal would indepedently track what samples are sent out from the IOC.

georgweiss, 2022-03-28:
This is channel access, and most probably not through gateway.

The weird plot in the screen shot I suspect also shows the case where the horizontal line is still there after reconnect (fixed in my recent PR). 

And yes, PV timestamp changes at 1Hz, here is a sample output from camonitor:
```
MEBT-010:PBI-FC-001:STAT3-MeanValueR 2022-03-28 15:02:51.042308 -0.00720035  
MEBT-010:PBI-FC-001:STAT3-MeanValueR 2022-03-28 15:02:51.042308 -0.00720035  
MEBT-010:PBI-FC-001:STAT3-MeanValueR 2022-03-28 15:02:52.042405 0.00522665  
MEBT-010:PBI-FC-001:STAT3-MeanValueR 2022-03-28 15:02:53.042209 0.00413015  
MEBT-010:PBI-FC-001:STAT3-MeanValueR 2022-03-28 15:02:54.041941 -0.00720035  
MEBT-010:PBI-FC-001:STAT3-MeanValueR 2022-03-28 15:02:55.042188 0.00157165  
MEBT-010:PBI-FC-001:STAT3-MeanValueR 2022-03-28 15:02:56.042177 0.0161916  
MEBT-010:PBI-FC-001:STAT3-MeanValueR 2022-03-28 15:02:57.042123 -0.00098685 
```

georgweiss, 2022-04-04:
The timestamps are "good", i.e. I have no reason to believe they are not what is provided by the IOC. They only arrive with a delay, which is seen using camonitor as well as in Phoebus. But the out-of-order side effect is entirely in Phoebus, or rather in jca.

Fixing insertion is in my view maybe not the optimal way forward, I'd be more interested to know why events arrive out-of-order or twice to `LiveSamples`.

kasemir, 2022-04-04:
Can you limit the issue to just live samples, using a channel that has no archived data?
Or does this only happen on channels that have archived data, where maybe the last sample in the history is newer than the next arriving sample from channel access?

georgweiss, 2022-04-04:
No sure what you are asking for. This is the strip chart widget, is consumes only live data from the IOC.

kasemir, 2022-04-04:
> This is the strip chart widget

Couldn't tell from the screenshot if it's a data browser or stripchart widget, and the underlying code is the same - except for fetching historic data.
Good, what you see is then limited to the live sample buffer, no worries about an overlap with historic data.

georgweiss, 2022-04-04:
Precisely, sorry for the confusion.

georgweiss, 2022-08-17:
To summarize.
There are two underlying issues resulting in the observed behavior:

1. PV disconnects and then reconnects after a (short) while. This case is handled by commit 3aeb70111b22b32384285271a2659dc67e7b4a73.
2. PV is accessed over a badly performing ca gateway. This case is not (and will not be?) fixed.

closed 2022-08-17 by georgweiss