ControlSystemStudio/phoebus #2203: Legacy rectangle widget with 'None' border style

tanviash, 2022-04-04:
We have some legacy rectangle widgets that have a 'None' border style but a non-zero border_width. These rectangles when opened with Display Builder have a zero line_width as enforced here https://github.com/ControlSystemStudio/phoebus/blob/master/app/display/model/src/main/java/org/csstudio/display/builder/model/widgets/OutlineSupport.java#L58.

Can we have the 'None' case to also follow other cases like 'Line' etc. so border_width doesn't get overlooked?

Also, border colors get overridden with NamedWidgetColors.TEXT for 'Line' and other border_styles but what if border color is set to some other color. Why not copy the border_color from BOY rectangle widget to here?

kasemir, 2022-04-06:
The BOY rectangle has "Line" as well as "Border", which is a bit odd:

[image]

The display builder only has one surrounding element for the rectangle, and that's the "Line".
It imports the width of a "Border" as the width of the "Line", and I agree that the color should then be imported as well instead of defaulting to black resp. the "Text" color:

[image]

But not sure what to do if legacy displays use both "Line" and "Border".
Maybe update the legacy displays to just use "Line", since that will work in both incarnations?

kasemir, 2022-04-06:
As for "border_style" being "None" yet the border_width is set, I'd consider that an error.

Update the legacy displays to set border style to line, or use "line_width" and "line_color", either way it should then work in both.