ControlSystemStudio/phoebus #2207: Elasticsearch version

xushuangs, 2022-04-11:
Hi all. I built a new linux test cluster, and the operating system was changed from Centos7 to Debian11. Debian11 no longer supports Elasticsearch6.x. Can Alarm Logging be used with elasticsearch 8.x? 
I am using phoebus 4.6.5.

georgweiss, 2022-04-11:
We're planning for a migration to Elasticsearch 8.x during Q2, but cannot say when this will be completed.

shroffk, 2022-04-11:
I have a temporary script to allow the use of elastic 7.x until we complete the migration to elastic 8. I wanted to add support for creating the templates automatically however this is not possible with the java client.
Please test it and let me know if it works and then I can open a MR

shroffk, 2022-04-13:
@xushuangs or @tanviash can you review

xushuangs, 2022-04-14:
I tryed to create the new index template through es7_mapping_definitions.sh. Then I can query the alarm data through curl and kibana.
sh phoebus-parent-4.6.5/services/alarm-logger/startup/es7_mapping_definitions.sh
{"acknowledged":true}{"acknowledged":true}{"acknowledged":true}(alarm-environment) 
curl -X GET 'http://localhost:9200/_template/*state*'
{"alarms_state_template":{"order":0,"index_patterns":["*_alarms_state*"],"settings":{"index":{"number_of_shards":"1"}},"mappings":{"properties":{"severity":{"type":"keyword"},"current_message":{"type":"keyword"},"mode":{"type":"keyword"},"message_time":{"format":"yyyy-MM-dd HH:mm:ss.SSS","type":"date"},"current_severity":{"type":"keyword"},"APPLICATION-ID":{"type":"text"},"pv":{"type":"keyword"},"latch":{"type":"boolean"},"time":{"format":"yyyy-MM-dd HH:mm:ss.SSS","type":"date"},"message":{"type":"keyword"},"config":{"type":"keyword"},"value":{"type":"keyword"}}},"aliases":{}}}
[image]

shroffk, 2022-04-15:
Great!! thanks for testing @xushuangs 
Can we close this Issue

closed 2022-04-20 by shroffk

kasemir, 2022-04-20:
To clarify, we now support ES 6 as well as ES 7, right?
ES 6 works with `mapping_definitions.sh`, ES 7 with `es7_mapping_definitions.sh`.

> Can Alarm Logging be used with elasticsearch 8.x?

Not at this time.