ControlSystemStudio/phoebus #2208: Phoebus application: cannot switch back from Full Screen Mode

pavel-ch, 2022-04-11:
I am not able to switch back from Full Screen mode (F11 or Menu entry) - testing on Windows. The only way seems to be manually modifying .memento.
Looking into `org.phoebus.ui.javafx.FullScreenAction` I think that it cannot work (onAction code referring always to the captured variable `full` created in constructor run time).
Maybe I am missing something.
Could someone verify?
Thanks in advance.

kasemir, 2022-04-11:
Yes, the FullScreenAction is created in https://github.com/ControlSystemStudio/phoebus/blob/d0f32c90084a290b4dbe7e02e66e15c376250f22/core/ui/src/main/java/org/phoebus/ui/application/PhoebusApplication.java#L551 to always enter full screen mode.

But that's just the initial menu content. In the following `menu.setOnShowing` https://github.com/ControlSystemStudio/phoebus/blob/d0f32c90084a290b4dbe7e02e66e15c376250f22/core/ui/src/main/java/org/phoebus/ui/application/PhoebusApplication.java#L556 it's updated to enter/exit full screen mode.

That worked fine until yet another `setOnShowing` was added in https://github.com/ControlSystemStudio/phoebus/blob/d0f32c90084a290b4dbe7e02e66e15c376250f22/core/ui/src/main/java/org/phoebus/ui/application/PhoebusApplication.java#L568 to populate the list of tabs.

--> Those two `setOnShowing` handlers need to be combined into one that both updates the fill screen entry and the tab list.

pavel-ch, 2022-04-11:
Ok thanks Kay. I assume whoever broke it will fix it too ;-)

georgweiss, 2022-04-11:
I will look into this.

georgweiss, 2022-04-12:
@pavel-ch , can we close this?

pavel-ch, 2022-04-12:
Thank you @georgweiss yes it works now, I am closing the ticket. I can confirm F11 does not work now (worked before) but that is no big issue for me.