ControlSystemStudio/phoebus #2211: Make date picker non-modal.

georgweiss, 2022-04-12:
I have a user request to make the date/time picker dialog in Data Browser non-modal. Though I am not sure why this is requested, I'd like to collect feed-back in this forum.

kasemir, 2022-04-12:
Blocking dialogs are generally bad for our type of operator interface. Selecting a start/end time in some plot should not prevent you from pushing a button in one of the other concurrently open displays. Especially not if the dialog happens to be hidden, so the complete application appears dead.
Should also be possible to look at two plots side by side, then adjust the start/end time in both of them by opening the time range GUI for both at the same time.

So making the date/time dialog non-modal is generally good.

Even better would be to somehow avoid a pop up dialog. When the dialog becomes non-modal, that creates the risk of the dialog staying open, being moved to the side, and then you forget which dialog is associated with what.

One approach would be a complete redesign of the start/end time selection. Don't open a pop-up dialog, but somehow re-do the GUI. For example, in the data browser property panel's "Time Axis" tab, replace the button that opens the dialog ...

[image]

.. with something that embeds those time selection elements into the "Time Axis" tab, so there's no dialog at all.
That's of course a larger effort. 

Maybe easier would be replacing the "pop up" with a "pop-over" type of dialog, like the ones used for font and color detail in the display editor:

[image]

They're clearly associated with the font/color/.. of a specific widget, there can be more than one.
But that's also more work, so maybe start with just keeping the existing dialog but make it non-modal.

georgweiss, 2022-04-12:
Thanks @kasemir for your input. Agree that a rework would be a desirable solution. Right now I can only find time to switch to non-modal.

shroffk, 2022-04-12:
currently the "time axis" tab view is rather empty, most of the times it is used to open the dialog... so adopting Kay's suggestion of populating it with the dialog elements makes a lot of sense.

secondly, there is the button in the plot toolbar which also opens the dialog... this is by far the most common way I have seen user get to the time range dialog. This dialog could be made into the "pop-over". I really like the idea of using "pop-over" dialogs more since in avoids the situation where you have orphaned non-modal dialogs hanging around on one of the 6 screens at our workstations.

Maybe we can add this to the list of projects for the code-a-thon

georgweiss, 2022-04-29:
A related use case is list selection UIs. On one hand we have the ListSelectionDialog - not using fxml - that is used by logbook editors. Then there is an fxml and associated controller used inside a PopOver in another logbook context (advanced search). 

So we have two implementations serving the exact same purpose, with almost exactly the same UI. 

I'd be happy to unify this in either of the following manners:
1) Modify ListSelectionDialog to use the fxml. Client code can choose to make it modal or non-modal.
2) Extend PopOver to a ListSelectionPopOver and delete ListSelectionDialog.