ControlSystemStudio/phoebus #2213: Use of TimeStampedProcessVariables

shroffk, 2022-04-12:
In the core data type along side the `ProcessVariable` there is another type `TimeStampedProcessVariable` this was intended for use where the selection would carry a pv name along with a point of time... a pv in a save set snapshot or a alarm log message or alarm leaf.

When opening other application it might be useful to open them with both the pv names and timestamps. The most immediate usecase is the databrowser... instead of opening it with a selected pv from the alarm log and the current time, it might be useful to open the databrowser around to the time of the alarm log message.

kasemir, 2022-04-12:
What do you suggest for "around the time"?

Using  saved point in time plus/minus the  `org.csstudio.trends.databrowser3/time_span` setting?
Or have a dedicated preference for that time span?
Or change TimeStampedProcessVariable from having just a point in time to holding a start..end time range?

shroffk, 2022-04-12:
| Using saved point in time plus/minus the org.csstudio.trends.databrowser3/time_span setting?

I have used this
```
instance.getModel().setTimerange(
                    TimeRelativeInterval.of(start.get().minus(time_span.dividedBy(2)),
                                            end.get().plus(time_span.dividedBy(2))));
```