ControlSystemStudio/phoebus #2238: Tab alignment in Tabs widget

Amadrath, 2022-05-06:
When changing the parameter "Direction" in the tabs widget, the text alignment of existing tab-labels does not change, but new tabs do. As the first tab can never be "new", its text alignment can not be changed for all tabs, which results in tab-labels with a mix of vertical and horizontal text alignments.
I would propose, to expand the properties of the tab widget to have one property "Label Text Alignment" with the options vertical and horizontal, and one property "Tab Position" with the options top, bottom, left and right. This would also add the option for right and bottom aligned tabs. The property "Tab height" should depend on text alignment, because the other dimension should be derived from the font size.

kasemir, 2022-05-06:
The tab text is roughly centered.
In the editor, "direction" can be changed.
Runtime changes of the direction as well as adding/removing tabs is not supported.
Related to runtime changes not being supported, this can sometimes lead to slightly misplaced labels while in the editor, see 'refreshHack' and 'sandbox.TabDemo'.

[image]

Changing direction:

[image]

Adding new tab in editor where label is initially poorly rendered:

[image]

"Fixes itself" after slight resize or saving & reopening or just waiting:

[image]

I'm hesitant to add options for individual label text alignment, tab position etc. because they would have to be maintained in the web runtime as well and then in the next display tool.
The basic design principle of the display builder is to have a tool for control system displays with longevity in mind, supporting a fairly generic display file format, see
https://github.com/ControlSystemStudio/phoebus/wiki/Display-Builder-Compatibility . Exact control of the pixel position of each tab's label is a very low priority

Amadrath, 2022-05-06:
I see my misunderstanding. I assumed the new tabs would keep their horizontally aligned text. But after a reload the text is aligned vertically, like the other tabs:
[image]–Reload–>[image]

Regrettably, that does not solve my initial problem. I want to have a tab widget, which is no more than 80px high and contains 3 widgets 20px high on top of each other. So there is no room for the tabs on top. With vertical text, there is no room on the side, either. But it would work with horizontal text, like this:
[image]

In order to maintain backwards compatibility, I would suggest to keep the existing parameter as is, and add an additional binary parameter "Flip Text Alignment", which changes the alignment of the text from the usual behaviour to the other one. If the parameter is missing in a bob-file, it defaults to the current behaviour. This would preserve backwards-compatibality

Amadrath, 2022-05-06:
I pondered that option, but I think, switching to a widget with a different function to compensate for the lack in flexibility of the initial widget, should be a last resort.
I might also simply put all those widgets in a separate display, called by a button.
While all those options would solve, or at least shift, my immediate problem, the usability of the tab widget remains limited. Thus I would still argue, that the tab widget should be extended, at least in the long run.

Amadrath, 2022-05-17:
I pondered that option, but I think, switching to a widget with a different function to compensate for the lack in flexibility of the initial widget, should be a last resort.
I ended up simply putting those widgets into a separate display, called by a button on the initial display. I basically split one crowded display into two similar, but less crowded, displays.
While all those options would solve, or at least shift, my immediate problem, the usability of the tab widget remains limited. Thus I would still argue, that the tab widget should be extended, in the long run.

Amadrath, 2022-05-17:
I ended up splitting the crowded display into two, one of which can be opened from the other. This allows me to hide some of the less used widgets.

kasemir, 2022-05-17:
Simplifying the display is always best. I won't find the time to add more options to the different types of tabs and maintain them in both the desktop and web runtime. If you do need full control over the exact look, consider building your own python display (see PyDM etc)

closed 2022-05-17 by kasemir