ControlSystemStudio/phoebus #2273: org.elasticsearch.client.RequestOptions$Builder NoSuchMethodError

daykin, 2022-05-24:
Hi,

not sure if this is the correct place to ask, but I'm trying to update everything to Elastic 8.2. If I start up the alarm logger after building with commit 9eb4d33a3 , I get the following runtime error: 

I think it might have something to do with spring-boot also looking for a different version of the old elasticsearch client, but I am not sure. Are there any maven experts here?

`Exception in thread "streams-Accelerator-alarm-cmd-03d2b427-675a-4f05-b1ee-489e96fb0161-StreamThread-2" java.lang.NoSuchMethodError: org.elasticsearch.client.RequestOptions$Builder.addHeader(Ljava/lang/String;Ljava/lang/String;)Lorg/elasticsearch/client/RequestOptions$Builder;
	at co.elastic.clients.transport.rest_client.RestClientOptions.initialOptions(RestClientOptions.java:179)
	at co.elastic.clients.transport.rest_client.RestClientTransport.<init>(RestClientTransport.java:103)
	at co.elastic.clients.transport.rest_client.RestClientTransport.<init>(RestClientTransport.java:107)
	at org.phoebus.alarm.logging.ElasticClientHelper.<init>(ElasticClientHelper.java:80)
	at org.phoebus.alarm.logging.ElasticClientHelper.getInstance(ElasticClientHelper.java:108)
	at org.phoebus.alarm.logging.AlarmCmdLogger.lambda$run$0(AlarmCmdLogger.java:126)
	at org.apache.kafka.streams.kstream.internals.KStreamPeek$KStreamPeekProcessor.process(KStreamPeek.java:42)
	at org.apache.kafka.streams.processor.internals.ProcessorNode$1.run(ProcessorNode.java:50)
	at org.apache.kafka.streams.processor.internals.ProcessorNode.runAndMeasureLatency(ProcessorNode.java:244)
	at org.apache.kafka.streams.processor.internals.ProcessorNode.process(ProcessorNode.java:133)
	at org.apache.kafka.streams.processor.internals.ProcessorContextImpl.forward(ProcessorContextImpl.java:143)
	at org.apache.kafka.streams.processor.internals.ProcessorContextImpl.forward(ProcessorContextImpl.java:126)
	at org.apache.kafka.streams.processor.internals.ProcessorContextImpl.forward(ProcessorContextImpl.java:90)
	at org.apache.kafka.streams.kstream.internals.KStreamTransform$KStreamTransformProcessor.process(KStreamTransform.java:59)
	at org.apache.kafka.streams.processor.internals.ProcessorNode$1.run(ProcessorNode.java:50)
	at org.apache.kafka.streams.processor.internals.ProcessorNode.runAndMeasureLatency(ProcessorNode.java:244)
	at org.apache.kafka.streams.processor.internals.ProcessorNode.process(ProcessorNode.java:133)
	at org.apache.kafka.streams.processor.internals.ProcessorContextImpl.forward(ProcessorContextImpl.java:143)
	at org.apache.kafka.streams.processor.internals.ProcessorContextImpl.forward(ProcessorContextImpl.java:126)
	at org.apache.kafka.streams.processor.internals.ProcessorContextImpl.forward(ProcessorContextImpl.java:90)
	at org.apache.kafka.streams.processor.internals.SourceNode.process(SourceNode.java:87)
	at org.apache.kafka.streams.processor.internals.StreamTask.process(StreamTask.java:288)
	at org.apache.kafka.streams.processor.internals.AssignedStreamsTasks.process(AssignedStreamsTasks.java:94)
	at org.apache.kafka.streams.processor.internals.TaskManager.process(TaskManager.java:409)
	at org.apache.kafka.streams.processor.internals.StreamThread.processAndMaybeCommit(StreamThread.java:957)
	at org.apache.kafka.streams.processor.internals.StreamThread.runOnce(StreamThread.java:832)
	at org.apache.kafka.streams.processor.internals.StreamThread.runLoop(StreamThread.java:767)
	at org.apache.kafka.streams.processor.internals.StreamThread.run(StreamThread.java:736)
2022-05-24 16:16:25 WARNING [org.apache.kafka.streams.KafkaStreams] stream-client [streams-Accelerator-alarm-messages-b1ccfcfa-5857-4547-b8a3-839565f7be92] All stream threads have died. The instance will be in error state and should be closed.
Exception in thread "streams-Accelerator-alarm-messages-b1ccfcfa-5857-4547-b8a3-839565f7be92-StreamThread-1" java.lang.NoSuchMethodError: org.elasticsearch.client.RequestOptions$Builder.addHeader(Ljava/lang/String;Ljava/lang/String;)Lorg/elasticsearch/client/RequestOptions$Builder;
	at co.elastic.clients.transport.rest_client.RestClientOptions.initialOptions(RestClientOptions.java:179)
	at co.elastic.clients.transport.rest_client.RestClientTransport.<init>(RestClientTransport.java:103)
	at co.elastic.clients.transport.rest_client.RestClientTransport.<init>(RestClientTransport.java:107)
	at org.phoebus.alarm.logging.ElasticClientHelper.<init>(ElasticClientHelper.java:80)
	at org.phoebus.alarm.logging.ElasticClientHelper.getInstance(ElasticClientHelper.java:108)
	at org.phoebus.alarm.logging.AlarmMessageLogger.lambda$processAlarmConfigurationStream$7(AlarmMessageLogger.java:230)
	at org.apache.kafka.streams.kstream.internals.KStreamPeek$KStreamPeekProcessor.process(KStreamPeek.java:42)
	at org.apache.kafka.streams.processor.internals.ProcessorNode$1.run(ProcessorNode.java:50)
	at org.apache.kafka.streams.processor.internals.ProcessorNode.runAndMeasureLatency(ProcessorNode.java:244)
	at org.apache.kafka.streams.processor.internals.ProcessorNode.process(ProcessorNode.java:133)
	at org.apache.kafka.streams.processor.internals.ProcessorContextImpl.forward(ProcessorContextImpl.java:143)
	at org.apache.kafka.streams.processor.internals.ProcessorContextImpl.forward(ProcessorContextImpl.java:126)
	at org.apache.kafka.streams.processor.internals.ProcessorContextImpl.forward(ProcessorContextImpl.java:90)
	at org.apache.kafka.streams.kstream.internals.KStreamTransform$KStreamTransformProcessor.process(KStreamTransform.java:59)
	at org.apache.kafka.streams.processor.internals.ProcessorNode$1.run(ProcessorNode.java:50)
	at org.apache.kafka.streams.processor.internals.ProcessorNode.runAndMeasureLatency(ProcessorNode.java:244)
	at org.apache.kafka.streams.processor.internals.ProcessorNode.process(ProcessorNode.java:133)
	at org.apache.kafka.streams.processor.internals.ProcessorContextImpl.forward(ProcessorContextImpl.java:143)
	at org.apache.kafka.streams.processor.internals.ProcessorContextImpl.forward(ProcessorContextImpl.java:126)
	at org.apache.kafka.streams.processor.internals.ProcessorContextImpl.forward(ProcessorContextImpl.java:90)
	at org.apache.kafka.streams.kstream.internals.KStreamPassThrough$KStreamPassThroughProcessor.process(KStreamPassThrough.java:33)
	at org.apache.kafka.streams.processor.internals.ProcessorNode$1.run(ProcessorNode.java:50)
	at org.apache.kafka.streams.processor.internals.ProcessorNode.runAndMeasureLatency(ProcessorNode.java:244)
	at org.apache.kafka.streams.processor.internals.ProcessorNode.process(ProcessorNode.java:133)
	at org.apache.kafka.streams.processor.internals.ProcessorContextImpl.forward(ProcessorContextImpl.java:143)
	at org.apache.kafka.streams.processor.internals.ProcessorContextImpl.forward(ProcessorContextImpl.java:122)
	at org.apache.kafka.streams.kstream.internals.KStreamBranch$KStreamBranchProcessor.process(KStreamBranch.java:48)
	at org.apache.kafka.streams.processor.internals.ProcessorNode$1.run(ProcessorNode.java:50)
	at org.apache.kafka.streams.processor.internals.ProcessorNode.runAndMeasureLatency(ProcessorNode.java:244)
	at org.apache.kafka.streams.processor.internals.ProcessorNode.process(ProcessorNode.java:133)
	at org.apache.kafka.streams.processor.internals.ProcessorContextImpl.forward(ProcessorContextImpl.java:143)
	at org.apache.kafka.streams.processor.internals.ProcessorContextImpl.forward(ProcessorContextImpl.java:126)
	at org.apache.kafka.streams.processor.internals.ProcessorContextImpl.forward(ProcessorContextImpl.java:90)
	at org.apache.kafka.streams.kstream.internals.KStreamMap$KStreamMapProcessor.process(KStreamMap.java:42)
	at org.apache.kafka.streams.processor.internals.ProcessorNode$1.run(ProcessorNode.java:50)
	at org.apache.kafka.streams.processor.internals.ProcessorNode.runAndMeasureLatency(ProcessorNode.java:244)
	at org.apache.kafka.streams.processor.internals.ProcessorNode.process(ProcessorNode.java:133)
	at org.apache.kafka.streams.processor.internals.ProcessorContextImpl.forward(ProcessorContextImpl.java:143)
	at org.apache.kafka.streams.processor.internals.ProcessorContextImpl.forward(ProcessorContextImpl.java:126)
	at org.apache.kafka.streams.processor.internals.ProcessorContextImpl.forward(ProcessorContextImpl.java:90)
	at org.apache.kafka.streams.kstream.internals.KStreamFilter$KStreamFilterProcessor.process(KStreamFilter.java:43)
	at org.apache.kafka.streams.processor.internals.ProcessorNode$1.run(ProcessorNode.java:50)
[... cut ...]
	at org.apache.kafka.streams.processor.internals.AssignedStreamsTasks.process(AssignedStreamsTasks.java:94)
	at org.apache.kafka.streams.processor.internals.TaskManager.process(TaskManager.java:409)
	at org.apache.kafka.streams.processor.internals.StreamThread.processAndMaybeCommit(StreamThread.java:957)
	at org.apache.kafka.streams.processor.internals.StreamThread.runOnce(StreamThread.java:832)
	at org.apache.kafka.streams.processor.internals.StreamThread.runLoop(StreamThread.java:767)
	at org.apache.kafka.streams.processor.internals.StreamThread.run(StreamThread.java:736)
`

kasemir, 2022-05-24:
I think ES8 is simply not supported at this time.
ES6 is, so is ES7, and we have instructions on how to copy existing data from ES6 to ES7, see https://github.com/ControlSystemStudio/epics-codeathon-2022/discussions/5
That recipe even copies data from ES6 or ES7 to ES8.

But ES8 is still not fully supported because it requires more dramatic changes. @shroffk would know more.

shroffk, 2022-05-24:
Hi Evan, thanks for continuing working on updating the alarm logger to es8...can you push your changes to a branch which I can look at.

kasemir, 2022-05-24:
My bad, you are actually working on moving to ES8, forgot that.
Still, Kunal would know more ;-)

shroffk, 2022-05-25:
https://github.com/ChannelFinder/ChannelFinderService/blob/elastic8/pom.xml

Here is the ChannelFinder elastic8 version - it is broadly similar to your pom. I wonder if the alarm logging ui which still depends on the old 6.8 might be the cause of the problem

I will do some testing today/tomorrow

daykin, 2022-06-01:
Go ahead!

shroffk, 2022-06-06:
I have fixed this on the [elastic-8-2-alarm-logger](https://github.com/ControlSystemStudio/phoebus/tree/elastic-8-2-alarm-logger) branch
@daykin can you test this

shroffk, 2022-08-10:
I have merged the solution
@daykin can we close this

closed 2024-06-12 by daykin