ControlSystemStudio/phoebus #2291: Detached windows' z index re-arranged

georgweiss, 2022-06-09:
I have a user reporting that overlapping (detached) windows will change z order after closing a model dialog. This was reported on Linux, but is easily reproduced on Mac OS.

Found the following (old) post: https://bugs.openjdk.org/browse/JDK-8089397, which is in line with my user's problem description.

Any reason to investigate this any further?

kasemir, 2022-06-09:
The bug report mentions "with an owner that is not the last opened window then when it closes a the last opened window will be brought to the front rather than the closed window's owner".

We have this commit to specifically set the owner of dialogs to the 'parent' window, with a comment "..to get the right dialog icons on Windows":

54fd0f668c68089c540f98ce84de09297778f8fe

If you can reproduce the issue, maybe try if removing that `initOwner` call helps, but overall this seems to be an internal issue over which we have little control.

kasemir, 2022-06-09:
If JDK 17 and higher indeed fix the problem, then that's a perfect solution.

georgweiss, 2022-06-09:
More testing is needed, of course, but I agree.
Unfortunately ESS cannot switch to Java 17 as target workstations' OS (CentOS 7) must be updated first.

georgweiss, 2022-06-09:
Actually... the ``initOwner`` fix will maintain the owner window in the front, but if multiple detached and overlapping windows are used, then those other windows may still be re-arranged. And this is unfortunately not solved by running under Java 17.

georgweiss, 2022-06-10:
Closing this as upon further testing I find Java 17 does not solve the issue originally reported by my user. JavaFX 19ea does not fix the issue either.

closed 2022-06-10 by georgweiss