ControlSystemStudio/phoebus #2296: ActionButton style

archman, 2022-06-13:
It seems that the ActionButton does not really look like a button when clicking it. Is it possible to make it look similar to a regular button that shows it gets pushed in and bounced out when pressing?

kasemir, 2022-06-24:
Try now

kasemir, 2022-06-27:
Turns out that to allow changing the background color of buttons, we had updated many of the associated styles (`-fx-color`, `-fx-outer-border`, `-fx-inner-border`, `-fx-background`) which are somewhat 'global' and also replace the coloring of the hover and pressed states.
In #2313 we now simply adjust the underlying `-fx-base`, so the hover and pressed indications remain functional, and the latter has been enhanced to be more obvious in