ControlSystemStudio/phoebus #2297: Multistate LED: Define states in class

mgotz, 2022-06-14:
Hi,
I would like to define the states and associated colors for the LED (Multi State) with a class. This does not seem possible. Similarly, you can't define the symbols for a Symbol Widget with a class. Is this by design or could it be added?

The background is that we have a lot of alarm state indicators with red-yellow-green lights in old edm panels. I would like to have a unified aproach to replace these in Phoebus. So, a  mulit state led with the class AlarmIndicator and/or a Symbol would be really useful.

kasemir, 2022-06-20:
I'll have to look at it, but my hunch is that array-typed properties are not properly handled by the 'class' mechanism.
Font of a label, on/off color of a plain LED, .. all the scalar types are fine, but array types like the N state colors of a multistate LED or the N image files of a symbol aren't handled.

kasemir, 2022-06-21:
Should now work for you

mgotz, 2022-06-23:
Thank you! This is great.

PR #2302 (2022-06-21): Display: Support 'class' for array/struct properties
Widget classes used to only support scalar properties like X position, Width, "On" color of a binary LED.

Now classes can be used with array/struct properties like the list of symbol files for a Symbol widget or the states of a MultiStateLED.

In the class editor, the whole array can be selected to be included in the class definition
[image]

When then editing displays, selecting the class applies the complete array to the widget
[image]

Similar example: Multi state LED with states set by class
[image]

Fixes #2297