ControlSystemStudio/phoebus #2300: General dependency vulnerabilities

pavel-ch, 2022-06-17:
As I am involved in our CyberSecurity programm, I checked at first stage Phoebus dependencies for known vulnerabilities. I am sure we have to solve these issues, it will be sooner or later necessary anyway.
The scan protocol of the OWASP dependency-check tool is attached. Please note that beside a number of HIGH severity dependencies, we have also several CRITICAL severity ones.
I know limitations of these SCM scans. Effective vulnerabilities are always dependent on HOW these components are used. But, anyway, I am convinced we should make some effort to minimize alerts, by for example upgrading dependencies to last versions without known vulnerabilities.
For the future, it is perhaps necessary to establish a project wide policy handling this issue.
Thank you for comments.
[Dependency-Check Report.pdf](https://github.com/ControlSystemStudio/phoebus/files/8926923/Dependency-Check.Report.pdf)

georgweiss, 2022-06-19:
As for Spring related dependencies (including tomcat), I and @shroffk are currently discussing updates across all service modules using Spring. 
Will also check postgres jdbc driver.

pavel-ch, 2022-11-29:
Jython 2.7.3 is available, with updated dependencies. Currently used Jython 2.7.2 generates some complains in dependency check, see below.
Can we think about an update?
Jython is currently in 3 poms: services/scan-server, dependencies/phoebus-target, app/display/runtime
The 2.7.3 is not tested by me so far, waiting for your opinion. There are some "improvements", no clue atm. what consequences can be to our scripts.
Thank you for considering @kasemir @shroffk 

> jython-standalone-2.7.2.jar (shaded: com.google.guava:guava:28.0-android) cpe:2.3:a:google:guava:28.0:*:*:*:*:*:*:* pkg:maven/com.google.guava/guava@28.0-android LOW 1 Highest 9 
> jython-standalone-2.7.2.jar (shaded: jline:jline:2.14.5)  pkg:maven/jline/jline@2.14.5   0  25 
> jython-standalone-2.7.2.jar (shaded: org.antlr:antlr-runtime:3.5.2)  pkg:maven/org.antlr/antlr-runtime@3.5.2   0  27 
> jython-standalone-2.7.2.jar (shaded: org.apache.commons:commons-compress:1.19) cpe:2.3:a:apache:commons_compress:1.19:*:*:*:*:*:*:* pkg:maven/org.apache.commons/commons-compress@1.19 HIGH 4 Highest 66 
>

kasemir, 2023-07-19:
Of those listed I've only used postgres (for the timescale DB support), where I think an update is fine.
Not directly calling guava or sqlite and not sure who could check if they're OK.

georgweiss, 2023-07-20:
@pavel-ch, I cannot detect any dependency to sqlite-jdbc. Output of ```mvn dependency:tree``` does not contain any sqlite references.

As for guava... the (J)Tango libs depend on it, but that I cannot evaluate/test. Logbook and channel finder have minimal dependency to guava so I assume update should be fine.

pavel-ch, 2023-07-20:
Thank you @kasemir @georgweiss ! I try postgres and guava. Sqlite is a local product dependency, sorry for mentioning that, I handle it locally.

pavel-ch, 2024-05-20:
At the moment, the Dependency Check reports vulnerabilities for SQL drivers:
postgresql.driver.version 42.6.0
mysql.version 8.0.30
Can I suggest version update to (latest):
postgresql.driver.version 42.6.2
mysql.version 8.4.0
?
Regarding testing, I am only able to test my use case, which works.
Thank you for an opinion @kasemir @shroffk @georgweiss !

georgweiss, 2024-05-20:
Not sure why you ask, this is what I see:
[image]

pavel-ch, 2024-05-20:
I depends on check tool, I use: http://jeremylong.github.io/DependencyCheck/dependency-check-cli/index.html
[image]
[image]
[image]

With vulnerabilities, I try to be on safe side... and update whenever any tool complains.

kasemir, 2024-05-21:
I thought there was an issue ticket about mysql/mariadb where a newer version means the server side must also be new, while the version we have can connect to older and newer servers.
But I can't find that now, and personally I don't rely on mysql nor Postgres, so I'm fine with updating the client versions.