ControlSystemStudio/phoebus #2304: Ability to close not currently selected tab

tynanford, 2022-06-22:
We got a feature request to add the ability to close a tab that is not the one currently in focus. 

For instance, there could be an "X" included in all tabs similar to an internet browser. And a middle mouse click would work on any tab. I would imagine this would best be created as a preference setting.

Any thoughts?

georgweiss, 2022-06-23:
Agree this would be useful.

kasemir, 2022-06-23:
That's already there:

Right-click on any tab, active or not, and the context menu contains "Close", "Close Others", "Close All".

tynanford, 2022-06-23:
It is an extra mouse click but that is true. 

This is probably a separate issue but maybe more the reason behind this request. In the case where you have 3 windows open, each with multiple tabs, closing a tab in a non-active window makes that window active. Which can be annoying if you have many different applications open all over the screen. 

Would it be difficult to have either the middle mouse click, 'X' button, or right-click -> 'Close' not make the window active? The use case is trying to quickly clean up clutter without losing the current window.

kasemir, 2022-06-23:
> Would it be difficult to have .. the .. 'X' button

Turns out no, see @2307
But clicking that still makes the overall window active, which you'll have to bring up with the JavaFX or maybe GTK people, its outside of what we can do.

closed 2022-06-23 by tynanford