ControlSystemStudio/phoebus #2322: PV's insertion with archive search

L-Roussel, 2022-07-07:
Hi,

Currently, We have two ways to add a pv on a graph (Data browser),
the first, with the option "add PV",
Or, the second, with the "archive search panel", by dragging a pv directly on the graph.

Using "add PV", the archive isn't defined, although it can be found in the "archive search panel".

The chosen archive for one PV should be the same either way.
In that case, we need to search if a pv already has an archive.

In the "AddPVorFormulaMenuItem" class, line 57 :
~~~
AddModelItemCommand.forPV(undo, model, dlg.getName(0), dlg.getScanPeriod(0), axis, null);
~~~
The archive is "null".
Maybe the archives search for a pv could be made right before that instruction.

I can implement that.

kasemir, 2022-07-07:
You may be overcomplicating this.
End users should not need to worry about the archive data source.
They just add a PV, and they see current as well as historic data.
If your end users first need to pick an archive data source, you did it wrong.

Ideally, you just have one archive system, so it's one URL and you're done.
Maybe you need to transition from an older system to a new one.
Then you can ideally again hide that behind one archive data source.
I believe the archive appliance for example can read from some other sources.
Or you "import" your older data into your current system.

Still, assume you painted yourself into a corner and need to support more than one archive data source.
Then you simply list them all in the `archives` setting, https://github.com/ControlSystemStudio/phoebus/blob/6424b43bf52cbadc7e041ae7bc6864a71d5dc0b0/app/databrowser/src/main/resources/databrowser_preferences.properties#L80
Whenever your user now adds a new channel, the data browser will query all of them.
It will drop those that don't contain the channel (https://github.com/ControlSystemStudio/phoebus/blob/6424b43bf52cbadc7e041ae7bc6864a71d5dc0b0/app/databrowser/src/main/resources/databrowser_preferences.properties#L93) and from now on only read from the source that actually knows the channel.
There is a tiny bit of overhead for one querying those sources that turn out to now have the channel, but I doubt that any end user will notice.

When you use the archive search panel, basically the same thing is happening.
Enter PV name, press "Search", and by default we search all data sources.
If the PV is found, it's listed with the data source. Drag that into a data browser, and it reads from that source. So in this case the "search" already determines where we need to read data, the data browser doesn't need to query all of them again.

The archive search panel does show the data source, which is mostly meant for system admins to debug the archive setup, for example to check that a channel isn't listed in multiple data sources. It's not meant to force end users to pick an archive, where they likely pick the wrong one, then complain that they don't find any data.

If you do want to change anything in here, one idea would be an option to by default hide the archive data sources in the search view, so most end users just see the PV name pattern and the list of PVs without any need to worry about the data source.

L-Roussel, 2022-07-07:
Thank you for your reply !

My issue was that the "Add PV" didn't use the DB stored values and so display only the real time values.

Actually, my settings had an error. I have two preferences for the archive :

~~~
# Suggested data servers
org.csstudio.trends.databrowser3/urls

# Default data sources for newly added channels
org.csstudio.trends.databrowser3/archives
~~~

Where the urls setting (used in archive search panel) was correct, the archives setting (used for "add pv") was inaccurate.

closed 2022-07-07 by L-Roussel