ControlSystemStudio/phoebus #2323: Log scale for Tank widget

archman, 2022-07-11:
It seems there is no property configuration for the LogScale of Tank widget? Or do you have an alternative solution to present a single scalar data in a bar plot?

kasemir, 2022-07-11:
The tank representation, https://github.com/ControlSystemStudio/phoebus/blob/master/app/display/representation-javafx/src/main/java/org/csstudio/display/builder/representation/javafx/widgets/TankRepresentation.java, is based on the https://github.com/ControlSystemStudio/phoebus/blob/master/app/rtplot/src/main/java/org/csstudio/javafx/rtplot/RTTank.java, and in there one can basically say `axis.setLogarithmic(true)` or `false`.
So the hard part is done, there's simply no property on the tank widget to configure that.
@tanviash , do you have somebody to add that?

kasemir, 2022-07-11:
Great, thanks

kasemir, 2022-07-20:
Implemented by #2327