ControlSystemStudio/phoebus #2325: core-pv ignore prefs

shroffk, 2022-07-14:
recently the core-pv was updated to ignore preferences and only use environment variables if the system property `jca.use_env` is set to true ( default)
however, this introduces a change in behaviour... previously the `var_array_supported` would be set based on the caj version (or pref) but now it is always false.
I don't know of an epics environment that overrides this... so we could either change the initialization of the var_array_supported to null, or in the if statement assocaited with the use `jca.use_env` we can set var_array_supported  to null.

kasemir, 2022-07-14:
> jca.use_env is set to true ( default)

It's true for PVWS, because the web app doesn't use the phoebus `-settings` command line option etc. but simply falls back to the EPICS_CA_.. environment variables. So this should only be a problem for PVWS, not the phoebus tools.

> in the if statement associated with the use jca.use_env we can set var_array_supported to null

Yes, can you fix this in PVWS? https://github.com/ornl-epics/pvws/blob/8074a95a8a25f747fd035b346a43fe5f81c5cf32/src/main/java/pvws/PVWebSocketContext.java#L57

kasemir, 2022-07-14:
Or change https://github.com/ControlSystemStudio/phoebus/blob/67a2511949a8985eb2d5c6b6bcd850903c0a79bf/core/pv/src/main/java/org/phoebus/pv/ca/JCA_Preferences.java#L52 to initialize

```
    private Boolean var_array_supported = null;
```

With phoebus, it's then soon set based on preferences. With PVWS, you get the auto-enablement based on the CA version, which is what you want for PVWS.

kasemir, 2022-07-14:
> so we could either change the initialization of the var_array_supported to null

Yes, that one seems best.
BTW, great that you figured this out!!

PR #2326 (2022-07-14): CA: Start var_array_supported=null ...
.. to remain compatible with PVXS

Fix #2325
commit 14c2496e (2022-07-14): CA: Start var_array_supported=null ...

.. to remain compatible with PVXS

Fix #2325
closed 2022-07-14 by shroffk