ControlSystemStudio/phoebus #2331: Missing core-security dependency in app-alarm-ui project

katysaintin, 2022-07-21:
org.phoebus.applications.alarm.ui.AlarmUI class in app-alarm-ui import
org.phoebus.security.authorization.AuthorizationService

So it's missing code-security dependency in the pom file

<dependency>
		<groupId>org.phoebus</groupId>
		<artifactId>core-security</artifactId>
		<version>4.6.10-SNAPSHOT</version>
</dependency>

shroffk, 2022-07-21:
Strange... the CI build should be failing.

kasemir, 2022-07-21:
alarm-ui depends on core-ui which depends on core-security.

Please, don't have me explain/defend maven ?!?!

closed 2022-07-21 by shroffk

kasemir, 2022-07-21:
Yes, the dependencies are hierarchical.
Now back with Eclipse plugins you actually needed to select if a dependency is "re-exported" to be transient. With maven I'm not sure if that's select-able, but it sure simplifies the dependencies if they can build a hierarchy.

kasemir, 2022-07-21:
I think I'm somewhat in between. With Eclipse plugins and now pom.xml it's nice that you don't always need to list everything. You list what you care about like core-ui or core-pv, and if that in turn depends on core-security, core-pva, core-formula, let alone jca, mqtt, ..., fine, it's automatically pulled in.

At the same time, I strongly feel about the overall project needing to know what's all included. Only depending on maven and having no clue what's in the final product is bad. What if you switch from Maven to another build tool, a tool that doesn't know the maven repository, with no idea about its dependency tree?

That's why we have https://github.com/ControlSystemStudio/phoebus/tree/master/dependencies which fetches all the dependencies into one folder and then you can fundamentally build any way you want, using ant or eclipse or a Makefile or shell script because the dependencies are all there.