ControlSystemStudio/phoebus #2333: IndexNameHelper

shroffk, 2022-07-21:
The Alarm Logging service has a nifty class which given a Instant generates an Index name based on preconfigured duration.
i.e.         
```
IndexNameHelper fromIndexNameHelper = new IndexNameHelper("test", "M", 1);
```
Would create a new index every month.
However, the way this service is implemented... the index names generated depend on the state of IndexNameHelper object when it was initialized and thus when you restart the service you might get varying results.

for example:
```
IndexNameHelper fromIndexNameHelper = new IndexNameHelper("test", "M", 4);
```
In the above case you want a new Index ever 4 months...if you are starting from Jan you would expect a new index on 
`test_2022-01-01` , `test_2022-05-01`, `test_2022-09-01`, ...

However if you restarted your server in Feb.. then you would get indices
`test_2022-01-01` , `test_2022-02-01`, `test_2022-06-01`, ...

Also, if you restarted your server in Feb, but did not use the Index name generator till March.. then you would get indices
`test_2022-01-01` , `test_2022-03-01`, `test_2022-07-01`, ...

While in the past this has not been an issue since we executed elastic queries against ALL indices, this might not be good long term solution... but in order to be able to execute index specific queries we need a way to reliably predict the index name.
This is made more difficult by the fact that the time when the IndexNameHelper is instantiated will produce different results even with the same date span unit and value.

Solutions:
1. the simplest solution might be to only allow 1Day, 1Month, 1Year.. this would completely avoid the unpredictability I described above.
2. reimplement the IndexNameHelper so that we can generate predictable outputs

At NSLS2 I am using a 1 Month setup and I am personally leaning towards solution 1 for its simplicity

@tanviash @georgweiss @kasemir What is the datespanunit and value you are using at your site and which of the above solutions do you prefer?

kasemir, 2022-07-21:
Default is one month, and that works fine.
I've run one setup with a 1 day config, and that was a stupid idea. Too many indices, slowing down the search as you collect several months worth of data.

I'd say only offering options 1D, 1M, 1Y is sufficient, and 1M may be the most practical option. 1D is bad. 1Y might be OK if you really don't have that much data.

tanviash, 2022-07-21:
Have been using 1M (default) and works great. Solution 1 seems fine.

georgweiss, 2022-07-22:
We will stick to 1M. Agree with @kasemir that 1D may not make sense.

Question is how to decide between 1M and 1Y. Should we consider to save us some application logic and only support 1M?

shroffk, 2022-07-22:
The overhead of supporting 1M and 1Y is pretty much nothing

For some of the beamlines where I am deploying the alarm services, they have a few dozen pv's and a few hundred events to log... so for that setup 1Y does make sense.

So I will make the agreed changes and commit it to the same branch on which elastic8 stuff is being done.

@tanviash can you test the new service with the client and the FRIB use cases.

tanviash, 2022-07-22:
@shroffk I already did, works fine. No issues with searching for all PVs within a node or an individual PV.

closed 2022-08-10 by shroffk