ControlSystemStudio/phoebus #2334: define a time interval for StripChart

L-Roussel, 2022-07-22:
Hi, 

I work on a stripchart widget and I want to display values over a time Interval, defined by two Timestamp, using a script.
Is it currently possible?

Maybe the time_range property value should be a TimeRelativeInterval instead of a String ?

kasemir, 2022-07-22:
> I want to ... use a script

See general script caveats in for example the https://controlssoftware.sns.ornl.gov/training/2022_USPAS/  display builder intro. In short, if you have to ask how to do a script, you'll likely have problems maintaining it as well, but here it is:

```
from org.csstudio.display.builder.representation.javafx.widgets import JFXBaseRepresentation
from java.time import Instant
from java.time.temporal import ChronoUnit

plot = JFXBaseRepresentation.getJFXNode(widget)
# You'll find that it's an RTTimePlot
print(plot.__class__.__name__)
# .. with a time range
print(plot.getXAxis().getValueRange())
start = Instant.now().minus(3, ChronoUnit.HOURS)
end = Instant.now().minus(1, ChronoUnit.HOURS)
plot.getXAxis().setValueRange(start, end)
```

kasemir, 2022-07-22:
Well, then you'll have to update the handling of the "time_range" property.
For example, create a new "start" and "end" property that's parsed like the start and end of the data browser.
Remove the "time_range" property, increment the version number of the stripchart widget and have it translate the "time_range" property from older version files into the new "start" and "end" properties.

kasemir, 2022-07-26:
Looks like you fixed this with #2338

Thanks!

closed 2022-07-26 by kasemir