ControlSystemStudio/phoebus #2335: app-alarm-ui Automated Actions Gui enhancement suggestion

katysaintin, 2022-07-22:
app-alarm-ui Automated Actions Gui enhancement suggestion

By replacing simple Textfield detail column by 2 columns :
- A combobox to select email or cmd
- A textfield to define the information (email or command)

The detail property will built as following option:info 
See the following screenshot to see the difference
[image]

I will push my code modification on my Fork https://github.com/katysaintin/phoebus

kasemir, 2022-07-22:
Since you already support `mailto:` and `cmd:`, can you also support `sevrpv:`?
See https://github.com/ControlSystemStudio/phoebus/blob/master/app/alarm/ui/doc/index.rst

katysaintin, 2022-07-22:
> Since you already support `mailto:` and `cmd:`, can you also support `sevrpv:`? See https://github.com/ControlSystemStudio/phoebus/blob/master/app/alarm/ui/doc/index.rst

I miss this one, Ok I will add to my suggestion.

katysaintin, 2022-07-22:
I have just add it (I use a combobox on enumeration)
[image]

katysaintin, 2022-07-22:
There is only a modification in one classe in app-alarm-ui . Do you think I can propose my modifcation as a pull request ?
https://github.com/katysaintin/phoebus/blob/master/app/alarm/ui/src/main/java/org/phoebus/applications/alarm/ui/tree/TitleDetailDelayTable.java

kasemir, 2022-07-22:
Yes, go ahead and create a PR

kasemir, 2022-07-26:
update had been merged

closed 2022-07-26 by kasemir