ControlSystemStudio/phoebus #2353: discussion - have complete settings.ini in target

tynanford, 2022-08-10:
I personally like to have a settings.ini file with the complete list of settings (everything on this webpage: https://control-system-studio.readthedocs.io/en/latest/preference_properties.html) in my CSS installation package. 

Then the site-specific things I want changed are uncommented and updated to whatever they may be. It also shows everything which is available to change and what the default is set to. Similar to how many config files in Linux under /etc look or for instance how Gitlab does their config: https://gitlab.com/gitlab-org/omnibus-gitlab/blob/master/files/gitlab-config-template/gitlab.rb.template

It's totally fine if no one else likes this. I'm just wondering what others think about it and if the phoebus build should create this full settings file

kasemir, 2022-08-10:
Way back in #4 I suggested that we use this type of preference access pattern:
```
Preferences prefs = Preferences.userNodeForPackage(...);
value = prefs.get(THE_NAME, default_value);
prefs.put(THE_NAME, value);
```

The idea was that you don't just read the current setting, but also write the setting back.
That way, as soon as your code once runs, the preferences are populated, initially with the default values.
So we would have a list of preferences.

After some discussions, we decided to NOT write the preferences.
You don't really know what's used and what their default values might be.
.. and you would not have any documentation about the use and meaning of preferences anyway.

Instead we have the agreement from https://control-system-studio.readthedocs.io/en/latest/preferences.html to put preferences into files named `*preferences.properties` with documenting comments, and then we locate all those files and combine them into the listing that's in the online help.

Yes, you could locate and parse those files into one long settings file.

What would you do with for example this?
```
# ----------------------------
# Package org.csstudio.archive
# ----------------------------

# RDB URL for archived data
#
# Oracle example
# url=jdbc:oracle:thin:user/password@//172.31.73.122:1521/prod
#
# PostgreSQL example
# url=jdbc:postgresql://localhost/archive
#
# MySQL example
url=jdbc:mysql://localhost/archive?rewriteBatchedStatements=true

# RDB user and password
# Some applications also provide command-line option to override.
user=archive
password=$archive
```

Extract just the settings into a format that can be used in the product, ready to set & un-comment selected lines?
```
#org.csstudio.archive/url=jdbc:mysql://localhost/archive?rewriteBatchedStatements=true
#org.csstudio.archive/user=archive
#org.csstudio.archive/password=$archive
```

Or also include the original descriptions:

```
# ----------------------------
# Package org.csstudio.archive
# ----------------------------

# RDB URL for archived data
#
# Oracle example
# url=jdbc:oracle:thin:user/password@//172.31.73.122:1521/prod
#
# PostgreSQL example
# url=jdbc:postgresql://localhost/archive
#
# MySQL example
#org.csstudio.archive/url=jdbc:mysql://localhost/archive?rewriteBatchedStatements=true

# RDB user and password
# Some applications also provide command-line option to override.
#org.csstudio.archive/user=archive
#org.csstudio.archive/password=$archive
```

tynanford, 2022-08-10:
I'd include the original descriptions to make the settings more clear personally. The file of course would be very long

georgweiss, 2022-08-11:
I'd be in favor of @tynanford 's idea. No matter how a site prepares a settings file, having the full list of settings - with descriptive comments - is helpful.

I think the comparison to Linux settings files is highly relevant.

kasemir, 2022-09-13:
Has somebody started to implement this?

There are about 80 files named `*preferences.properties` in the source tree.
A python script could locate them all, parse the initial `# Package ...` and then basically copy the content, replacing `setting = value` with `# the.package.name/setting = value`.
The result would be one long file similar to the example shown above.
I think this would cover 99% of all use cases.

It does have the drawback that it would include settings that may not be applicable.
For example, it would include the settings for save-and-restore and the scan client, but your site specific GUI product might not include them, and they also don't apply when you configure the alarm server.

Creating a settings template that only lists settings as they apply for a product (basic phoebus-product GUI, site specific product, alarm server, ...) would require to first determine what's included.
One approach could be to do this at the end of the build process:
Check all the `product/lib/*.jar` files for `*preferences.properties`, parse & transform them into one large preference setting template for the product.

tynanford, 2022-09-13:
My current "solution" here is hacking on this python function to get a settings file with all the settings (no comments)

Here is that file:
[full_settings.txt](https://github.com/ControlSystemStudio/phoebus/files/9560131/full_settings.txt)

This is the function I hacked on:
https://github.com/ControlSystemStudio/phoebus/blob/master/docs/source/conf.py#L312

```
# Create preference_properties.rst by listing
# content of all **/*preferences.properties files.
def createPreferenceAppendix(root):
    # Locate all files
    pref_files = dict()
    for (dirpath, dirnames, filenames) in walk(root):
        for filename in filenames:
            if filename.endswith("preferences.properties"):
                pref_file = path.join(dirpath, filename)
                # Use only files from sources, not those copied to target/classes
                if "target/classes" in pref_file:
                    continue
                pack = get_package(pref_file)
                pref_files[pack] = pref_file

    with open('full_settings.ini', 'w') as out:
        out.write("# Preferences Listings\n\n")
        for pack in sorted(pref_files.keys()):
            pref_file = pref_files[pack]
            with open(pref_file) as prefs:
                for line in prefs:
                    if line.startswith("# Package "):
                        package_str = line[10:].strip()
                        out.write("### " + package_str + "\n\n")
                    elif "=" in line and line[0] != "#":
                        line = line.strip()
                        line = line.replace("\\n", "")
                        line = line.replace("\\r", "")
                        out.write("# " + package_str + "/" + line + "\n")
            out.write("\n")

        out.write("\n")
```

tynanford, 2022-09-13:
I'd be happy to work on a cleaned up python function if this is something people are interested in having. Just would need help on where to place the file and when/how to get it to run during build process. 

Also guidance on if we want all settings in file, settings based on build, settings and all comments, etc.

kasemir, 2022-09-13:
My leaning is a `create_settings_template.py {product}` that searches all the `{product}/lib/*.jar` for `*preferences.properties`. So the "Locate all files" section becomes a little longer since it needs to deal with ZIP files (jar files), but rest is as you have it. It creates a template that has all the applicable settings, and the tool can easily be used on any local, site-specific product.  ... after the built, but would be easy enough to run this in a local build pipeline just before zipping the result up for distribution, so a site specific product can include the template.

tynanford, 2022-09-13:
sounds like a good plan to me, thanks. Might be a week or two before the PR arrives. Maybe I'll have an optional argument for enabling comments in the file as well

kasemir, 2022-09-13:
Terrific!  Yes, I'd include comments. Basically copy the files, capturing the "Package" and replacing `setting = value` with `# the.package.name/setting = value`.

closed 2022-09-16 by tynanford