ControlSystemStudio/phoebus #2368: inaccurate archived data for MySQL

L-Roussel, 2022-09-01:
Hi, 

I've recently noticed some problem with the archived values on MySQL.
The smpl_time values are rounded causing an error of one second.
For example, let's say we have a value registered the 2001/01/01 at 10:00:00.6
the value for the nanoseconds column is 600.000.000;
but the smpl_time is rounded at 2001/01/01 10:00:01.

It's caused by the Timestamp type in MySQL, that round to the second by default. 
Postgres and Oracle have not that problem, because the Timestamp type has by default more precision for them (6 digits for Postgres).

Some solution in the base :
- let the timestamp have more precision "Timestamp(1)" will be sufficient, but others applications can use the current version of the archive.
- add a trigger before insertion

in Phoebus :
- We trunk the variable that contains the date before using it in the query
- We trunk the date during the insertion query (maybe the best idea)

in mysql.dbd :
~~~
DROP TABLE IF EXISTS sample;
CREATE TABLE IF NOT EXISTS sample
(
   channel_id INT UNSIGNED NOT NULL                       COMMENT 'Channel ID',
   smpl_time TIMESTAMP NOT NULL                           COMMENT 'Time stamp',
   nanosecs INT UNSIGNED NOT NULL                         COMMENT 'Nanoseconds of time stamp',
   severity_id INT UNSIGNED NOT NULL                      COMMENT 'Severity ID',
   status_id INT UNSIGNED NOT NULL                        COMMENT 'Status ID',
   num_val INT NULL                                       COMMENT 'Integer value or null',
   float_val REAL NULL                                    COMMENT 'Float value or null',
   str_val VARCHAR(120) NULL                              COMMENT 'String value or null',
   datatype CHAR(1) CHARACTER SET binary NULL DEFAULT ' ' COMMENT 'Array element count, 1 for scalar',
   array_val LONGBLOB  NULL                               COMMENT 'Array data',
   
   FOREIGN KEY (channel_id) REFERENCES channel (channel_id) ON DELETE CASCADE,
   FOREIGN KEY (severity_id) REFERENCES severity (severity_id) ON DELETE CASCADE,
   FOREIGN KEY (status_id) REFERENCES status (status_id) ON DELETE CASCADE
);
~~~

kasemir, 2022-09-01:
Looks like you have already investigated this quite well!

We write the timestamp in https://github.com/ControlSystemStudio/phoebus/blob/522ec8619747feb980dfb73921847d00143114d7/services/archive-engine/src/main/java/org/csstudio/archive/writer/rdb/RDBArchiveWriter.java#L478-L484 
We read it back in https://github.com/ControlSystemStudio/phoebus/blob/522ec8619747feb980dfb73921847d00143114d7/app/databrowser/src/main/java/org/phoebus/archive/reader/rdb/AbstractRDBValueIterator.java#L179-L184

Can you confirm that the issue is with writing the data? `PreparedStatement.setTimestamp` will already round the seconds 'up'?
Then the solution should be somewhat like this:

```
// MySQL nanosecs
if (dialect == Dialect.MySQL  ||  dialect == Dialect.PostgreSQL)
{
    // MySQL stores nanos in designated column
    insert_xx.setInt(6, stamp.getNanos());
    // Clear nanos in timestamp to prevent following
    // setTimestamp(2, stamp) from rounding seconds based on nanos 
     stamp.setNanos(0);
}
insert_xx.setInt(1, channel.getId());
insert_xx.setTimestamp(2, stamp);
insert_xx.setInt(3, severity);
insert_xx.setInt(4, status.getId());
```

Just check that there are actually a few more places in the RDBArchiveWriter that call setTimestamp, so they would all need to be fixed like this.

L-Roussel, 2022-09-01:
I confirm, but i think PreparedStatement.setTimestamp just put the entire timestamp (example : '2022-07-19 14:23:53.404862658'),
and the db round it to obtain the nearest value. 

Anyway, I think your change will work, I'll try that.

kasemir, 2022-09-01:
We actually did a similar thing in the Eclipse-based codebase, https://github.com/ControlSystemStudio/cs-studio/commit/eb1e3902edd1ed795b3911e258cc4db9698beb85
This one will be a nanosecond more efficient because it doesn't create another truncated copy of the time stamp but simply truncates the one time stamp. But maybe keeping the `stamp` unchanged also has its benefit.

kasemir, 2022-09-08:
Fixed with PR

closed 2022-09-08 by kasemir