ControlSystemStudio/phoebus #2376: CSSTUDIO-1338: ScaledSlider and Scrollbar Widgets Reflect Incorrect PV Value When Dragged Outside Limits

cjenkscybercom, 2022-09-08:
(This mirrors CSSTUDIO-1338)

When the ScaledSlider or Scrollbar widgets are:

- connected to a PV with DRVL/DRVH limits
- configured s.t. those limits can be bypassed (the "Limits From PV" property is false)

If you try to set the PV to a value outside its DRVL/DRVH limits, then the widget displays the new value outside the limits (rather than the actual value, which is ofc set to the DRVL/DRVH values). For example, if the DRVL is 5 and you use the ScaledSlider widget to set the value to -20, then the ScaledSlider displays the value as -20 (even though it is actually set to 5). 

This is not the case for other widgets such as Spinner and TextEntry; if you e.g. attempt the above in the TextEntry widget then the value stays at 5 as expected.

Please see the attached db and bob files. Run the DB file locally via:
```
softIocPVA -d CSSTUDIO-1338.db
```

cjenkscybercom, 2022-09-08:
@kasemir Wanted to sanity check with you on this before putting in work into changing this behavior for those widgets? Thanks!

kasemir, 2022-09-08:
This is somewhat tricky. When we write a value to the PV, we don't know if the PV will accept that value. So the general idea is: User provides some value, we write it, but we then display the "old" value until we receive an update from the PV.
See detail in TextEntry submit, 
https://github.com/ControlSystemStudio/phoebus/blob/d37701e7443d333a11156f18cb82ee27add5bd95/app/display/representation-javafx/src/main/java/org/csstudio/display/builder/representation/javafx/widgets/TextEntryRepresentation.java#L234-L266

For the Text Entry, that works great:

1) Have some old value 4
2) User enters 100, we sent 100 to PV but restore "4"
3) PV returns actual value. Maybe 100, maybe 10 because of DRVH=10. Whatever, we display it

Since we only _schedule_ the redraw to restore "4" and the PV update arrives so quickly, the end user typically sees "4", enters "100" and sees "100" (or "10") right away.

With a slider, it's harder, because we don't want this:

1) Value is 4
2) User slides to "100", we send 100, but move slider back to 4 since as far as we know that's the current value
3) PV returns actual value, may be 100 or 10 or ..., so we move slider there.

That would look ridiculous as the slider jumps back and forth.
So we need to somehow allow the user to move the slider and send those values to the PV, when when the user is _done_, update to what the PV tells us. The basic ScrollBar tracks mouse clicks and an `active` state to accomplish that, see https://github.com/ControlSystemStudio/phoebus/blob/master/app/display/representation-javafx/src/main/java/org/csstudio/display/builder/representation/javafx/widgets/ScrollBarRepresentation.java, but you might have found cases where it doesn't work as intended.

kasemir, 2022-09-08:
Ideas for possible approaches that have been used in various display tools are basically to detect when the user is "done" via mouse-button-release and then force an update from the most recent PV value, or start/restart a time with every user interaction and some time after the last interaction force an update to the most recent PV value.

You might miss the mouse-button-release if the mouse is simply moved outside of the widget, which suggests a timer. But what time? User might press the button and then move the slider slowly for a minute. Should the slider keep jumping back every few seconds?

georgweiss, 2022-09-08:
> So we need to somehow allow the user to move the slider and send those values to the PV, when when the user is done, update to what the PV tells us

Based on this I'd say that in the example of the scaled slider is does indeed not work as intended. Attached video shows an example.

https://user-images.githubusercontent.com/35602960/189190972-55089cb4-559b-4d41-8838-df61b8e688b3.mov

kasemir, 2022-09-08:
Right, it's missing something like scheduling an update to the most recent PV value after `MOUSE_RELEASED`.

cjenkscybercom, 2022-09-09:
Thank you for your comments! Indeed, this sounds like something isn't working as expected. Good point about the update cycle; I'll have to play around with this on the mouse-up event, and see if the slider still bounces around uncomfortably.

cjenkscybercom, 2022-09-09:
@kasemir One thing I'm wondering is if the PV writes should be happening continuously, or if we should only request the write once the user releases their mouse drag?

I'm not familiar with the use case of this as much, so I'm not sure why writes would be fired continuously / as the value changes? (though yes it is throttled)

kasemir, 2022-09-09:
We definitely cannot wait until release. Operators will press on a slider and move it around, button held down and moving, while watching how the system responds.
In BOY, there was code to throttle the writes down, so we're not necessarily writing more than say 10 values per second, but that's a separate issue. Certainly have to write _some_ values while moving the slider.

kasemir, 2022-09-09:
Also, imagine the same display being open in the control room on two computers.
Operator on computer A moves the slider -->  Slider on display B should move by itself because slider in display A is writing to PV, PV sends updates, slider on B reflects those updates.

kasemir, 2022-09-09:
The slider is the most complicated widget in terms of reading and writing.
We do write as the slider moves. We don't want to jump around while moving, so we have some "active" flag to ignore PV updates while moving. While not moving, we do reflect updates from the PV.
What's missing is a forced update at the end of an operator move.

cjenkscybercom, 2022-09-12:
Took me a bit longer than I'd be willing to admit to realize I needed to mark the dirty_value before calling toolkit.scheduleUpdate. This works, and unless you drag the slider really fast there isn't any jitter / jumping back and forth. I'm not sure if/how that jitter should be addressed besides introducing some way to delay scheduled updates in the Toolkit task queue?

kasemir, 2022-09-12:
Looking at the CSSTUDIO-1338 branch, you basically added a refresh on mouse release, which should handle most cases.

Can you check for ways in which we could miss the mouse release?
Like press down, then move mouse out of the window, or press escape, or press Alt-Tab to switch applications, anything you can come up with that would cause us to miss a mouse release, so then we're stuck with the widget still thinking we're in mouse-tracking mode? Maybe add a listener to lost focus and tread that just like mouse-release to assert that we're not stuck in mouse-tracking mode.

Minor quirk: Looks like your editor changes specific imports into wildcard `import javafx.scene.layout.*` when there are several imports from the same package. Try to change that editor setting so it won't do that, since it's better to know exactly where everything originates. Way back when we had to type the imports it made sense to type less by using wildcard imports, but now that the editor automatically handles the imports, we don't save any keystrokes by using wildcards but do loose knowledge of what exactly is used.

cjenkscybercom, 2022-09-14:
> Looking at the CSSTUDIO-1338 branch, you basically added a refresh on mouse release, which should handle most cases.
> 
> Can you check for ways in which we could miss the mouse release? Like press down, then move mouse out of the window, or press escape, or press Alt-Tab to switch applications, anything you can come up with that would cause us to miss a mouse release, so then we're stuck with the widget still thinking we're in mouse-tracking mode? Maybe add a listener to lost focus and treat that just like mouse-release to assert that we're not stuck in mouse-tracking mode.
> 
> Minor quirk: Looks like your editor changes specific imports into wildcard `import javafx.scene.layout.*` when there are several imports from the same package. Try to change that editor setting so it won't do that, since it's better to know exactly where everything originates. Way back when we had to type the imports it made sense to type less by using wildcard imports, but now that the editor automatically handles the imports, we don't save any keystrokes by using wildcards but do loose knowledge of what exactly is used.

So far not seeing anything coming up on missing mouse release, at least on Mac.

Thought I fixed the import issue in IntelliJ...Think I finally nailed it down per updated info on https://stackoverflow.com/a/3348855/19043338, hopefully those don't escape back into commits again.

kasemir, 2022-09-14:
> So far not seeing anything coming up on missing mouse release, at least on Mac.

Great, and thanks for checking!
So go ahead, create a PR

PR #2385 (2022-09-14): Csstudio 1338: ScaledSlider and Scrollbar Widgets Reflect Incorrect PV Value When Dragged Outside Limits
Fixes #2376
closed 2022-09-14 by kasemir