ControlSystemStudio/phoebus #2377: closeAllTabsKeyCombination Breaks UI Tests

lorenzo-gomez-windhover, 2022-09-09:
Hi there,

I was looking at some UI testing, in particular [SplitDockTestUI](https://github.com/ControlSystemStudio/phoebus/blob/master/core/ui/src/test/java/org/phoebus/ui/docking/SplitDockTestUI.java), and now it faiils after `closeAllTabsKeyCombination` was added in 670900dc58a38bd7906c4c33db228690405e0878. And it's because there is a null pointer exception if one tries to configure a DockStage in isolation, which what the unit test tries to do.

I don't want to change a unit test to "make it pass" because that would defeat the whole purpose of the test. The first thing that comes to mind is adding a null-check to the code in DockStage. Something like the following:

```
        // This is to handle a keyboard shortcut that closes all tabs in all windows.
        stage.getScene().setOnKeyReleased(ke -> {
            if(PhoebusApplication.closeAllTabsKeyCombination !=null) 
            {
                if(PhoebusApplication.closeAllTabsKeyCombination.match(ke))
                {
                    PhoebusApplication.closeAllTabs();
                }
            }
        });
```

Perhaps a more elegant solution would be to wrap `closeAllTabsKeyCombination` in some kind `Optional` object.

Not writing a pull request on this right now since I want to make sure doing this would not have any unwanted side effects since this is a global and wanted to hear some feedback from the devs who know more about `closeAllTabsKeyCombination` than me.

georgweiss, 2022-09-10:
I would maybe just move

```
if(PlatformInfo.is_mac_os_x){
            closeAllTabsKeyCombination = new KeyCodeCombination(KeyCode.W, KeyCombination.SHIFT_DOWN, KeyCodeCombination.SHORTCUT_DOWN);
        }
        else{
            closeAllTabsKeyCombination = new KeyCodeCombination(KeyCode.F4, KeyCombination.SHIFT_DOWN, KeyCombination.CONTROL_DOWN);
        }
```
to a static initializer in ```PhoebusApplication```

georgweiss, 2022-09-12:
See #2378