ControlSystemStudio/phoebus #2408: CSSTUDIO-1363: Changes to Embedded Widget Aren't Reflected to Parent without Refresh

cjenkscybercom, 2022-10-05:
[This issue mirrors CSSTUDIO-1363]

Putting this here to sanity-check on how this could be implemented, after some investigation I've done. 

This can be replicated in the following way (see the attached zip of .bob files):

1. Load up the parent in the editor and viewer
2. Load up the child in the editor
3. Change the child in some way; either resize the Label widget, or change the text in it (maybe make the fox leap over a playful instead of lazy dog 🤷‍♀️ )
4. Save the child file
5. Navigate to the parent; in the editor or viewer, changes to the child aren't reflected. To make them visible, refresh the parent display. 

Looking into the implementation details of the EmbeddedDisplayRepresentation etc, it doesn't appear at first simple as hooking into e.g. a property and using it. 

I **can** do this, and it'll 'work' (with flickering); however this causes a cycle that eventually freezes/crashes phoebus. 

Ideally I'd want to be able to compare the old and new display runtimes, or old and new files. But in debug-inspecting the listeners, I'm finding either the old value, or new value, but not both, is present so I'm not quite sure I could compare them (or frankly, even *should* as that could be expensive with many embedded embedded displays). 

So, maybe I could check the hash of the file instead? However the file property only cares about the file name rather than hash, so I'd need to add a new property. I would guess that is doable, though it won't necessarily e.g. catch changes to embedded displays within an embedded display.

But all that said, I'm not able to trigger those listeners as I expect e.g. by saving the child with the parent open in another tab; I'd expect the child runtime display to have "changed" when I make a change to it and save, but that listener isn't triggered anyways.

Either way, I'm curious @kasemir if you have any thoughts regarding this request from a technical perspective. I'm not sure also quite on the use-case myself, it seems the intent was to build it around pull-based refreshes (user clicks refresh) rather than push-based? (Phoebus pushes update to refresh display)

[parent-and-child.zip](https://github.com/ControlSystemStudio/phoebus/files/9713694/parent-and-child.zip)

kasemir, 2022-10-10:
Yes, you have to reload a display to get updated embedded content.
In 30 years of using EPICS display managers that support embedded displays, that has not been an operational issue.
In fact, we added caching to the `ModelResourceUtil` so that displays fetched from http links are not read every time.
Going the other way and checking them all the time in a background thread for changes would go against the basic idea of reducing the network traffic.
So I'd really want to avoid changes to the runtime behavior.

We can, however, reconsider the behavior while *editing* displays.
For example, you could install some global notification mechanism where saving a display in the editor triggers all EmbeddedDisplayRepresentation instances to call their fileChanged listener.
The TabsRepresentation and TemplateInstanceRepresentation could refresh as well.
So we don't change the runtime behavior. Widgets that embed other displays don't actively read their content over and over, they don't compute hashsums or check file dates etc. Instead, we limit this to editing. The editor must be open within the same instances of CS-Studio, and the editor triggers a refresh on saving its file.

cjenkscybercom, 2022-10-11:
Thanks @kasemir ! 

Considering this and the effort required to change this (plus that in introducing this to the editor, this would create the expectation for this to be in the runtime and that has its own problems as you've pointed out), I'm going to go ahead and close this as a "won't do" for now.

closed 2022-10-11 by cjenkscybercom

kasemir, 2022-10-11:
Well, I wouldn't push it completely off the table.

We don't want the widget runtime to actively keep checking for changes all the time.

But we could add a global "reload" trigger to the widget runtime. As long as it's not fired, there's no cost at all.
The editor would fire it on "save", which certainly is better than having to manually reload all the time while editing, and it's limited to just the instance where you're editing files.

Later, we could even come up with a way to bind this reload trigger to a PV, so that you can fire a reload on all running instances in the control system on for example a maintenance day. That way, there's no longer a need to tell all operators: If you're looking at that page, please reload because we added some new bugs.