ControlSystemStudio/phoebus #2415: Timestamp validation between ca and pva

georgweiss, 2022-10-13:
I have a PV running on a (I guess) freshly restarted IOC and that upon ```camonitor``` will respond:
```
ISrc-010:Vac-VVMC-01100:FlwSPS <undefined> 0 UDF INVALID
```
The time stamp of the PV when decoded in Phoebus is the "default" EPICS (1990-01-01T:00:00:00).

Specifying ca as default protocol in Phoebus will construct a VDouble for this where the timestamp is invalid, while using pva will construct a VDouble where the timestamp is valid (since getEpochSeconds returns > 0). Is this intentional? 

The difference in implementations have a side effect on the DataBrowser. Using pva causes the sample in question to be added at time 1990-01-01T00:00:00. This in turn will make all historic samples from the archiver invisible, even if the plot is saved with a time window that does not include "now" since the first received live value will trigger this side effect.

Not sure where to fix this, and first question to address I think is the difference in timestamp validation between ca and pva.

kasemir, 2022-10-13:
I don't see a difference when I run this with `softIocPVA -d /path/to/file.db`:
```
record(ai, "demo")
{
}
```

[image]

Either way, the time stamp is zero, which then turned into the local time zone is something close to January 1, 1990.

georgweiss, 2022-10-14:
The difference in behavior I am pointing out is whether a time stamp is evaluated as valid or not. For instance:
in ```org.csstudio.trends.databrowser3.model.PVSamples.addLiveSample()``` we have:
```
 if (! Time.timeOf(value).isValid())
            value = VTypeHelper.transformTimestampToNow(value);
        addLiveSample(new PlotSample(Messages.LiveData, value));
```
For ca ``` Time.timeOf(value).isValid()``` will return ```false```, while for pva we get ```true```. The consequence is obvious: for ca the timestamp is converted to "now", while for pva it is not. Hence the side effect described earlier.

kasemir, 2022-10-14:
I think I found it.
For PVA, the seconds we get in Decoders.decodeTime are 631152000, not zero, because PVA uses the more common 1970 epoch.
I'll add a check for that.

kasemir, 2022-10-14:
This will mean that a time stamp of UTC 1/1/1990 00:00:00.0000000 which in principle would be a valid PVA time stamp will be considered invalid because that's what IOCs report based on 1990 being the Channel Access epoch.

I assume that will only be a problem for people who take PVA on a time travel back into 1989..1990.

closed 2022-10-17 by georgweiss