ControlSystemStudio/phoebus #2420: Data Browser : Legend PV selection

GDH-ISIS, 2022-10-14:
The Data Browser is a very useful tool to visualize data. To help with the visualization, when the plot legend is selected to be visible, it would be very useful to select the PVs displayed using the legend (do not remove the PVs configured). For example

- Simple mouse/select on a legend entry to display a single PV
- use CTRL/select on legend entries to add/remove PVs displayed
- use SHIFT/select for an array of PVs
- PVs not selected on the legend are greyed (leaving them available to be selected)
- Simple option to display all/grey out all.

This would allow users to select from an array of configured PVs very quickly without the add/subtract PV gui, allowing additional flexibility to browse the available data.

kasemir, 2022-10-14:
Sounds great. When you implement this, please also update the web runtime, https://github.com/ornl-epics/dbwr
In fact its data browser implementation is currently quite limited and we'd like to get the web version closer to the desktop version.

GDH-ISIS, 2022-10-14:
This was just an idea.

closed 2022-10-14 by GDH-ISIS