ControlSystemStudio/phoebus #2424: Save and Restore not included in nightly builds

UrbanBobek, 2022-10-17:
Hi all,

I downloaded the latest nightly build of CS-Studio (Phoebus) from the SNS Controls [website](https://controlssoftware.sns.ornl.gov/css_phoebus/) and followed the instructions to install it.  While applications such as Alarm Area Panel, Data Browser, Scan Editor, etc., were included, Save and Restore was not. If I build Phoebus from source as described in the README, Save and Restore is included in Phoebus. Is this intentional and if so, is there a way to include Save and Restore without compiling from source?

Thanks,
Urban

kasemir, 2022-10-17:
The products that you can download from the SNS website are examples for site-specific products. Since we don't use save-restore at the SNS, we don't include it. See https://github.com/ControlSystemStudio/phoebus/tree/master/phoebus-product for creating your own product where you can include or omit exactly what you want.

georgweiss, 2022-10-17:
When build from this repository, save&restore is included.
Please note that we are working on a major upgrade of save&restore that will bring some simplifications both in terms of maintenance and deployment. This is expected to be in place before the end of 2022.

UrbanBobek, 2022-11-04:
Hi, 
after a discussing with my colleagues I just wanted to double check this.
> The products that you can download from the SNS website are examples for site-specific products. Since we don't use save-restore at the SNS, we don't include it. 

On the site I linked to there are download links for ORNL/SNS/HFIR sites product and a General (non-ORNL/SNS/HFIR) product.
In my question about Save and Restore not being included in the nightly build I was referring to the General build. Are there any plans to include it in the future General nightly builds?

Either way we will wait for the upgrade.

Thanks!

georgweiss, 2022-11-04:
We will merge the save&restore updates soon (during November 2022).

That said, building a generic product is quite straight forward. With JVM (version 11 or newer) and Maven in place, it's as simple as 
```
mvn clean install
``` 

Build artefacts (zip or tar.gz) contain all the resources needed, and will also contain launch scripts.
As for the various services - including save&restore - these are built as executable jar files and found in the target directory of each product module. Please consult the READMEs found in the service modules (save&restore README to be updated) as the services have dependencies to things like Elasticsearch and Kafka.

kasemir, 2022-11-04:
> On the site I linked to there are download links for ORNL/SNS/HFIR sites product and a General (non-ORNL/SNS/HFIR) product.

If you refer to https://controlssoftware.sns.ornl.gov/css_phoebus/, that's still an SNS/ORNL site, where I build products for SNS. That includes one product for use on the actual machine, and "a generic CS-Studio product", but still using those tools that we generally use at SNS, compiled with `ant`, using the java version that we find appropriate at the time and use locally.

You can easily run the `mvn` build yourself, using the java version that you find appropriate. I'm not sure if anybody else out there is currently running that `mvn` build and publishes the results. On https://github.com/ControlSystemStudio/phoebus, we mention that you can run "mvn clean install" yourself, but we're not linking to any binaries.

In the end, you likely want to build your own site-specific product, see https://github.com/ControlSystemStudio/phoebus/tree/master/phoebus-product