ControlSystemStudio/phoebus #2435: Shared layouts via url

GDH-ISIS, 2022-10-28:
We are interested in potentially sharing common fixed layouts for numerous support teams around our facility and noted that the .bob Phoebus screens can be accessed via a web service/url. This has proven helpful in centralising our phoebus screens (using nginx). I was wondering whether the phoebus layouts (memento files) can be shared in a similar way (nginx has been configured to show the files in a directory via a url). I am not sure whether phoebus is able to see such files though (via a url entered for "org.phoebus.ui/layout_dir" option in settings.ini). Having the local layouts available per user allows for customisation and is very useful.

kasemir, 2022-10-28:
We implement this in a site specific launcher, i.e., the script that starts phoebus.

The saved layouts are stored in the `phoebus.user` location, which defaults to `$HOME/.phobus` but you can configure it via `-Dphoebus.user=...` at startup.

In our launcher script, we have code that copies "shared" layouts from a common location like `/home/controls/shared_layouts` to that `phoebus.user` location.
For example, a file like `/home/controls/shared_layouts/Overview.memento` is copied into the `$HOME/.phobus` folder, so every user can now select "Overview" from the shared layouts.

You might even create a file  `/home/controls/shared_layouts/.memento` which then gets copied into the `$HOME/.phobus` folder, and since the ".memento" is the file used as a default on startup, that means all users will always start out with that layout when they start css. That can be useful on beamlines because you can tell users who go lost to just exit and restart css to get back to the default layout. Or you decide _not_ to create a file `/home/controls/shared_layouts/.memento`  so users always start back up with what they had last open when they closed css.

Bottom line, you can do a lot of this with simply moving the desired memento files around in your site specific launcher script.

kasemir, 2022-10-28:
As for storing mementos on a web server, you then simply add something like `curl http://....` to your startup script to download the latest shared memento from the web.

GDH-ISIS, 2022-10-31:
@kasemir, many thanks for the prompt replay and the information. I appreciate that this was more of a support question.
The script option looks like the answer - in particular the curl idea looks neat and tidy.

closed 2022-10-31 by GDH-ISIS