ControlSystemStudio/phoebus #2436: Capturing keyboard inputs

bsobhani, 2022-10-28:
Is it possible to capture keyboard inputs, and perform actions when they are pressed? For example, identify that the up arrow is pressed, and then write to a PV accordingly. This can be used for example to move an ROI up or down using the keypad.

Does CSS have the capability to do this? I don't see it in the rules so if it exists I would imagine it would be in the scripting.

kasemir, 2022-10-28:
Some widgets like the Spinner or Scaled Slider, typically those widgets that have an "Increment" property, support cursor up/down or left/write to adjust the value while that widget has the focus.

bsobhani, 2022-10-29:
Thanks. I was envisioning something like giving the user the full up/down/left/right experience (i.e. without clicking between widgets) for moving an ROI. But something like this might work too - I hadn't thought of using the spinner widget.