ControlSystemStudio/phoebus #2438: Autoscaling visible region only

bsobhani, 2022-10-29:
For the image widget I can enable autoscale and then it will set the min/max of the colorscale to the min/max of the raw image data.

However, if I zoom into a region (by double clicking on the numbers on the axes and changing them), the autoscale is still for the entire image, not the zoomed in part.

Is there a way to make it so that autoscale uses only the data in the zoomed in region?

kasemir, 2022-11-04:
Misunderstood "autoscale". There's the scaling of the value range, and the scaling of the axes.
The scaling of the axes was broken. With "limits from PV" checked to auto-scale the axes, one could no longer zoom/pan, which is now fixed.
The auto-scale of the values still uses the complete image...

kasemir, 2022-11-04:
Changing the value autoscale would be a big deal because we currently always draw the complete image and the zoom/pan is just a matter of then copying that section from the image drawn in background to the display.

Besides, changing the colors when zooming in/out seems wrong because you're changing the data.

On beamlines here, the autoscale tends to be done differently anyway. The detector histogram images tend to contain some spots with very high counts. If those determine the maximum of the color scale, the rest looks pretty much black. So the color range is determined by computing the range of values within some region of interest outside of those peaks. I believe that's done by using area detector stats plugins. That min/max range is then applied, so the peaks may actually overflow the value range, but the "rest" now becomes visible.

bsobhani, 2022-11-04:
This is not for a detector but for a 2D image which shows the evolution of an incremental d-space plot over time. They want to zoom in and observe the movement of the various peaks over time, so the only important thing is for there to be some contrast between the maximum and minimum values of the specific peak they are zooming into, so that they can see the peak.

If there is some way in the Phoebus python scripting interface to access the data that is being displayed, it seems like it would be possible for me to add a button that calculates the max/min values of this, and set the color scale to these. But I do not know if there is a way to access this data.

kasemir, 2022-11-04:
MIght be easiest if you update https://github.com/ControlSystemStudio/phoebus/blob/master/app/rtplot/src/main/java/org/csstudio/javafx/rtplot/internal/ImagePlot.java to first check the 'zoomed' area, then determine the value range in there, then draw.

kasemir, 2022-11-07:
Now have an update that autoscales the value.

Example image where there are just a few value pixels for example in the bottom row, second from left portion of the detector which drive the overall autoscaling and thus result in the rest being quite featureless:

[image]

Zooming in on another part of the detector, autoscaling just the visible part makes those pixels more vibrant:

[image]

At the same time, this means that the color no longer represents any absolute value. Values for colors keep changing with the zoom/pan.

bsobhani, 2022-11-08:
I just tried it out and I confirm it does this. Thank you!

PR #2448 (2022-11-04): Fix image zoom/pan with "limits from PV"
Only update 'full' image size when it's actually changed.

Fixes #2438 where "limits from PV" would effectively prevent zoom/pan since each PV update resets not only the overall image size in user coordinates but also the axes.

Now "limits from PV" allows zoom/pan until the image is again resized.
commit 8920919e (2022-11-04): Only update 'full' image size when it's actually changed

Fixes #2438 where "limits from PV" would effectively prevent zoom/pan
since each PV update resets not only the overall image size in user
coordinates but also the axes.

Now "limits from PV" allows zoom/pan until the image is again resized.
closed 2022-11-08 by bsobhani