ControlSystemStudio/phoebus #2442: Filebrowser app: drop selection on root

georgweiss, 2022-11-03:
I have a user request (for the Filebrowser app) to enable dropping a selection onto the area below the TreeView. The intention is to support a  move/copy operation to the root of the view. This type of gesture is possible in the MacOS Finder, but to my knowledge it may not make sense on Windows Explorer or a Linux file manager.

While I have a working solution, I'd like to collect feed-back on the feasibility of the request. After all, users may expect dropping a selection to the empty space below the TreeView to be a no-op.

kasemir, 2022-11-03:
Drag & drop can be a little iffy. It's prone to delays/snags/freezes. For moving files I often prefer the command line...

Right now, if you want to move a file from a sub-folder to the root of the file browser, you need to drop it onto an existing file at the root level. That works, but is a little confusing. You might expect that this replaces the existing file with the moved one, while it will in fact move the dropped file to that level in the hierarchy. Allowing a drop "below" the listed files would be more intuitive, so adding that sounds good to me.

georgweiss, 2022-11-03:
Yes, the OS native tools are much better at file operations...

closed 2022-11-04 by georgweiss