ControlSystemStudio/phoebus #2459: PVA: cant write to custom struct field

kingspride, 2022-11-11:
is probably related to #2410 and #2457

when writing to a field (type agnostic) in a custom PVA struct, phoebus tilts with 
```
Caused by: java.lang.Exception: Element 'TPID' of 'TPID.value' is not a structure
	at org.epics.pva.data.PVAStructure.locate(PVAStructure.java:339)
	at org.epics.pva.client.PutRequest.encodeRequest(PutRequest.java:120)
	at org.epics.pva.common.TCPHandler.sender(TCPHandler.java:186)
	at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1136)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:635)
	at java.base/java.lang.Thread.run(Thread.java:833)
```

structure: 
```
structure
     boolean ITLK false
     boolean TRIG false
     boolean PWR false 
     boolean STAT false
     boolean TPID true
     double TAMP 28.05516609992207
     double TLIM 49.0
     double STG1_SWR 1.0
     double STG1_FWD 0.0
     double STG1_REV 0.0
     double STG2_SWR 1.0
     double STG2_FWD 0.0
     double STG2_REV 0.0
```

I guess the fix will be as easy as in the mentioned issues from yesterday ;)

thanks a lot!
- W

kasemir, 2022-11-11:
It's different from the handling of boolean types.

In a generic client like CS-Studio, we're trying to handle normative types. We look for a "value". If you have some arbitrary structure, we see if there's a "value" element in there and then display that. We also look for a "timeStamp" and assume that's the, well, time stamp, an "alarm" for alarm info and so on.

Similar, when asked to write, we try to write to a "value" element. Since "TPID" has no "value", you get that error when trying to write.

With PVAccess, you can now have custom structures, but that means you create the server to generate that custom data, and you very likely also need to create the client to really handle that custom data. We have some limited support for peeking into a custom structure to fetch just one element, but I'm sure this will always run into some issues. For starters, there won't be time stamp, alarm, display, ... information. Note how the preliminary `info(group,...` mechanism for combining records on the IOC into a custom data structure basically assembles several normative types, so each element has a "value", "timeStamp", etc. as a sub-element, but I don't think it supports writing to those.
We simply can't have a general tool support any custom data structure.

Still, I'll see if we can try and simply write `pva://some_name/some_field` if there's no ``some_name/some_field/value` below as we require for now.

kasemir, 2022-11-11:
How would you write to that custom structure outside of CS-Studio?

Do you have an example for using `pvput` to a field of your custom struct?

hjunkes, 2022-11-11:
pvput FHIFEL:3gigamp_1 TLIM=50

pvget FHIFEL:3gigamp_1
FHIFEL:3gigamp_1 structure
    boolean ITLK false
    boolean TRIG false
    boolean PWR false
    boolean STAT false
    boolean TPID true
    double TAMP 28.1572
    double TLIM 50
    double STG1_SWR 1
    double STG1_FWD 0
    double STG1_REV 0
    double STG2_SWR 1
    double STG2_FWD 0
    double STG2_REV 0

kasemir, 2022-11-11:
OK. If you try just `pvput FHIFEL:3gigamp_1 50` you get an error that mentiones "no value", right?

hjunkes, 2022-11-11:
Jupp,

Error: Structure has no .value

kasemir, 2022-11-11:
Good. So we'll try to do the same: Assume "value" as a default, but fall back to just the subfield

PR #2460 (2022-11-11): PVA write "value" vs. custom element
Write support always used the `.value` subfield. Now it can be used with `pva://custom_struct/some_field` and will then write just to `some_field`, not requiring `some_field.value`.

Still checking for `.value` to support the expected case of a "normative type".

Simplified the the PVNameHelper to just track one "request" since read and write requests are now the same.

Tested with plain `softIocPVA` type records (read and write), updated `BoolDemo` (read and write) and area detector image (read-only).

Fixes #2459
closed 2022-11-11 by kasemir