ControlSystemStudio/phoebus #2468: Feature request : markers for all axis

agaget, 2022-11-18:
Hello,
Love the markers features for the X/Y Chart widget. Should be nice to have the possibility to choose the axis.
For example here 2 more markers for Y markers. We use that kind of markers to compute automatically the X axis markers and detect width of beam by itself.

[image]

Cheers

agaget, 2022-11-18:
Is there any reward for the 100th opened issue ?

kasemir, 2022-11-18:
To put a (permanent) marker on an axis, we tend to add a PV like "=4.2" to get one at value 4.2.

As for rewards, you get a virtual inflatable washing machine for solving the 100th issue.

georgweiss, 2023-02-06:
We have a related (same?) request: to add/move/delete markers on the X axis using a toolbar button.