ControlSystemStudio/phoebus #2485: Update Application: Timestamp behavior

MichaelRitzert, 2022-12-16:
I'm implementing an update client that pulls the update ZIP file from the package repository of a gitlab project.

The principle is working fine, but while implementing the details, I found a problem that should also effect the existing code: This line
`current_version = TimestampFormats.parse(prefs.get("current_version"));`
uses TimestampFormats to parse the current_version string defined in the installed package.
However. TimestampFormats uses
`zone = ZoneId.systemDefault()`
to define the timezone used to interpret the time.

This means that it's not possible to reliably pass a timestamp from the build process (in my case the timestamp of the latest commit) to the application via settings.ini.

The current implementation somewhat hides the problem by defining the new configuration entry by itself, again with TimestampFormats.

My proposal would be to include the ISO8601 format as an allowed option in TimestampFormats:
```
diff --git a/core/util/src/main/java/org/phoebus/util/time/TimestampFormats.java b/core/util/src/main/java/org/phoebus/util/time/TimestampFormats.java
index 3ca7b1e37..ff84d386f 100644
--- a/core/util/src/main/java/org/phoebus/util/time/TimestampFormats.java
+++ b/core/util/src/main/java/org/phoebus/util/time/TimestampFormats.java
@@ -96,7 +96,8 @@ public class TimestampFormats
         TimestampFormats.MILLI_FORMAT,
         TimestampFormats.SECONDS_FORMAT,
         TimestampFormats.DATETIME_FORMAT,
-        TimestampFormats.DATE_FORMAT
+        TimestampFormats.DATE_FORMAT,
+        DateTimeFormatter.ISO_INSTANT
     };
 
     /** Try to parse text as absolute date, time
```

Then I could use a timezone-aware format in my autogenerated settings.ini.

georgweiss, 2023-02-07:
@MichaelRitzert , can we close this?

MichaelRitzert, 2023-02-07:
Yes, I'll close.

closed 2023-02-07 by MichaelRitzert