ControlSystemStudio/phoebus #2502: Change phoebus user (credentials management ?)

agaget, 2023-01-10:
Is there a possibility to change the user using phoebus on a session ?  I'd like to be able to change the user to get more or less authorization from the CAGateway (or later for the pvacces dedicated gateway can't remind the name). 

Would like to change from user "test" to user "Mickey" for example
When I do that manually `ssh -XY Mickey@localhost phoebus` or `ssh -XY test@localhost phoebus`, it works the gateway gives me the correct right, but I'd love to do that more smoothly throught GUI

[image]

For now when I click on the user button I got the credentials management, but I'm not sure it is dedicated to this behaviour.
@georgweiss I saw that it's you that develop that part.

Thanks

kasemir, 2023-01-10:
> Is there a possibility to change the user using phoebus on a session ? 

No.

Channel Access doesn't allow changing the user. It uses who's logged in. Basically the same with PV Access.

In the Eclipse-based version of CSS, we did allow "logging in" as a different user for access to alarms or other things besides channel access, but the main thing people want is really Channel Access. Plus the details of how to authenticate (LDAP? Custom certificates? Different operating systems?..) just became too much to maintain.

georgweiss, 2023-01-11:
Regarding the credentials management module...

The purpose of this is to support "caching" of credentials. For instance, the Olog logbook requires user to specify credentials as the remote logbook service will check if the user is authorized to add a log entry. In order to not prompt user for credentials each time, the credentials manager's API can be used to save and later retrieve the credentials.

The credentials manager is generic in the sense that it may store credentials for arbitrary use cases (aka "scopes").

agaget, 2023-01-11:
Ok, thanks guys, it's clearer for me now. I close the issue.

I perfectly understand the difficulty to maintain, then how do you solve this issue ? (if you have such authorization matter).
Thanks

closed 2023-01-11 by agaget

kasemir, 2023-01-11:
> Then how do you solve this issue ?

Well, you need to log in as whoever you need to be. On a Mac and also some Linux setups you can simply log in as another user and then switch between the various logged in sessions. So it's very straight forward, but you end up with different desktops.

On Linux, you may create some "run as other user" script that basically calls "sudo" to change the user, then start the tool. That used to be quite easy, but might now require calling `xauth` to get/set some cookie that X11 uses to permit access to the display, since the tool is now running as a user ID that's different from the owner of the display.

Using `ssh other_user@localhost` may be an easier way to open the tool within the same desktop, but I think it's slower because now you'll run all the X11 traffic via the ssh tunnel instead of directly to the display.

agaget, 2023-01-11:
Thanks for you answer
 chatGPT proposed me something I will try that ;)