ControlSystemStudio/phoebus #2503: Issue in accessing pvaccess PV on different subnet on windows standalone client installation

joshikk, 2023-01-11:
I have downloaded the overnight build of Phoebus for Window release and Configured the remote IOC server IP address (on a different subnet) using settings.ini
```org.phoebus.pv.ca/epics_ca_auto_addr_list=true
org.phoebus.pv.ca/addr_list=127.0.0.1 IPofRemoteSystem
org.phoebus.pv.pva/epics_pva_addr_list=127.0.0.1 IPofRemoteSystem
org.phoebus.pv.pva/epics_pva_auto_addr_list=true
```
I am able to access the CA PV on the remote IOC, however I am unable to access the PVA PV with this installation
If the remote IOC is on the same subnet there is no issue in accessing the PVA PV's.

Simultaneously, I have cloned a phoebus git on Ubuntu and using the above setting I am able to access the CA and PVAccess PV's without any issues.

I am wondering if I require to configure/install anything on Windows system to access PVAccess PV's, or is it a bug??

kasemir, 2023-01-19:
Update: Checking the search command sent from both the windows and the ubuntu client host, they look the same.

But while the ubuntu host sends the message to UDP `certain_ip:5076` and then gets a reply from `certain_ip:other_port`, the windows host does not see a reply.

This suggests that some firewall is blocking the message from the windows client to `certain_ip`, or a firewall, maybe on the windows host, is open for port 5076 only, so the reply from `other_port` is not seen?

joshikk, 2023-01-20:
Dear Kay,
Thank you for the exhaustive reply.
As you correctly identified, the server replies on another port. Hence the
windows firewall is blocking the same.
However, I am wondering, how could I communicate if the PVA server was on
the same subnet?

Once I disabled the windows firewall I am able to access it.
I was wondering if there is a gateway application which can work for both
CA and PVA?

Regards,
Kuldeep

kasemir, 2023-01-20:
> Once I disabled the windows firewall I am able to access it.

Excellent!

>  how could I communicate if the PVA server was on the same subnet?

Must be some detail in the firewall setting where it allows any UDP packets from the local subnet but not from other subnets.

> .. server replies on another port. Hence the windows firewall is blocking ..

That's an issue with the original C++ implementation of the PVA server, https://github.com/epics-base/pvAccessCPP/issues/159
It renders it unusable when crossing subnets and firewalls.

The new C++ implementation, https://github.com/mdavidsaver/pvxs, and the new PVA gateway that's based on it, https://mdavidsaver.github.io/p4p/gw.html,  stick to the configured UDP server port which you can then open in the firewall.

> I was wondering if there is a gateway application which can work for both CA and PVA?

CA and PVA are different protocols, and they each have their own gateway.
But you can easily run both a CA and a PVA gateway on the same host.
For CA, you often actually need to run multiple instances to get symmetric or reverse connections. With the new PVA gateway, one instance can handle multiple connections.

joshikk, 2023-01-21:
Thank you for your help. It was a windows firewall issue

closed 2023-01-21 by joshikk