ControlSystemStudio/phoebus #2506: Feature Request: Make waveform PVs in 1D table widgets "just work™"

daykin, 2023-01-13:
Current behavior:

If I plug a waveform PV into the pv_name field of a Table widget, it is interpreted as "Row 1, Column 1 is set to an array of numbers". I can work around this with PVUtil.GetStringArray(), casting to an array of 1-element arrays.

Desired behavior:

I plug in a waveform PV, and each element in the waveform is resolved to its own cell.

daykin, 2023-01-26:
I will be attending the March codeathon at Diamond. I'll plan on addressing this.

kasemir, 2023-01-26:
The `VTable` type that the table widget uses internally is column-oriented, https://github.com/epics-base/epicsCoreJava/blob/master/epics-vtype/vtype/src/main/java/org/epics/vtype/VTable.java

This would make it quite easy to turn a waveform PV, for which you already have a `VxxxArray` type, into a `VTable` with that data as the one and only column. Can be done in a zero-copy implementation that just calls `VxxxArray.getData()` from the wrapping `VTable.getColumnData(0)`.

I'm afraid, however, that most users would naively expect to see a single-row table with one column per array element, i.e., a "horizontal" table instead of a "vertical" one. That'd require creating a `VTable` with many columns and copying each array element into a single-element array for that column...

daykin, 2023-03-07:
@kasemir Is the second commit (a8b76) what you had in mind? I had done this another way then saw your comment.

closed 2023-03-15 by daykin