ControlSystemStudio/phoebus #2508: Databrowser "Statistics" indicate wrong "Mean"

kasemir, 2023-01-17:
Local user found a situation where the "Mean" appears to be outside the "Min.." .. "Max Value":

[image]

kasemir, 2023-01-17:
Hard to read, but the red trace shows Min -0.11..., Max 0.12... with Mean 7.18....

georgweiss, 2023-01-17:
Could it be that there is an exponent for the mean value?

georgweiss, 2023-01-17:
Of course we could make the columns wider, I'll look into it.

kasemir, 2023-01-17:
Yes, "7.18E-2" = 0.07 would be an explanation. Unclear if that instance is still running to verify

kasemir, 2023-01-17:
User was able to change the column width on that instance, and the Mean was indeed "7.18...E-4".

Yes, having the column widths will the width of the pane would have prevented this misunderstanding, though you'll then still be able to make the window size too small to show all values, so there's no perfect solution.

Anyway, nothing severely wrong here.

kasemir, 2023-01-23:
Yes, thanks!

closed 2023-01-23 by kasemir